UZ-SLAMLab/ORB_SLAM3 #94: Does the method support Global shutter or Rolling shutter?

yike-chen, 2020-09-04:
Thanks for your great work.
My camera is ZED 2, does it need the additional operation to support?

jjgr3496, 2020-09-04:
Currently ORB-SLAM3 only supports Global Shutter cameras. Unfortunately, adapting ORB-SLAM3 to rolling shutter cameras is not straightforward and we are not planning to do it.

closed 2020-09-04 by jjgr3496

Shkunn, 2021-03-23:
Hello ! I've experimented the same probleme which is :

```
not IMU meas
no pose update
not enough acceleration
no pose update
not enough acceleration
no pose update
not enough acceleration
```
So @yike-chen I've read all your messages and figured that the probleme was that ZED2 is a Rolling Shutter camera and ORD-SLAM3 only supports Global Shutter cameras as @jjgr3496 said...

It's been 6 months since you've posted your message so I'd like to know if you had been working on this issue since then ?

Appreciate it !