UZ-SLAMLab/ORB_SLAM3 #96: Running ROS Euroc example

andriskuciks, 2020-09-07:
Hello! I'm trying to run ROS example but whenever I load vocabulary I get following output:
```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 3.85272

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
double free or corruption (out)
Aborted (core dumped)
```
Tried the solution mentioned [here](https://github.com/raulmur/ORB_SLAM2/issues/341), but still no luck. Any ideas whats causing this?

jjgr3496, 2020-09-07:
At a first glance, I would say this is a compilation issue due to using a different compiler (or same compiler but different versions) to build g2o, DBoW2 and ORB-SLAM3 libraries. Please, check that you are using the same exact compiler (including the version) for building everything.

However I am not sure about the cause of this error so I would ask you to find in which line of code you are getting this error.

andriskuciks, 2020-09-08:
Hi @jjgr3496 !
So I did some debugging and discovered that this error is from `Tracking.cc` line 794: `Tbc = node.mat();` causing the program to crash. 
I checked `CMakeCache.txt` file for g2o, DBoW2 and ORB-SLAM3 and it appeared that all of these libraries are using the same compiler. At least all the option with 'compiler' in its name were the same. Maybe I done it wrong

andriskuciks, 2020-09-11:
Didn't really figure out what was causing the `core dump` error. But I installed virtual box with clean Ubuntu 18.04, followed the install instructions and now everything wokrs. Tho there's a new problem. When I run for example _MH_03_medium.bag_ dataset ROS stereo, both ORB_SLAM2 and ORB_SLAM3 can't keep up with the incoming frames and looses track pretty fast. It tries to re localize but after a short while its lost again.  Is it because in MH_03_medium.bag theres a lot of movement or the lack of computing power? 
Heres is the `rosrun` output:
```
orbslam@orbslam:~/ORB_SLAM3$ rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/EuRoC.yaml true

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 3.85272

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0.014865543, -0.99988091, 0.004140297, -0.021640146;
 0.99955726, 0.014967213, 0.02571553, -0.064676985;
 -0.025774436, 0.0037561883, 0.99966073, 0.0098107308;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00017 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.002 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.not IMU meas
First KF:0; Map init KF:0
New Map created with 244 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
LM: Active map reset recieved
LM: Active map reset, waiting...
IMU in Map 0 is initialized
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1
[... 294 lines cut ...]
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 215
mnInitialFrameId = 213
208 Frames set to lost
not IMU meas
First KF:134; Map init KF:121
New Map created with 662 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
RESETING FRAME!!!
Segmentation fault (core dumped)
```

closed 2020-09-11 by andriskuciks