UZ-SLAMLab/ORB_SLAM3 #97: "Undefined error to symbol error, error adding symbols, DSO missing from command line", while building for ROS

BeardyMan37, 2020-09-07:
`./build_ros.sh 
Building ROS nodes
build exists, deleting
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
[... 1614 lines cut ...]
      |                                                                          ~~~~~~^~~~~~~~~~~~
In file included from /usr/local/include/eigen3/Eigen/Core:344,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/ImuTypes.h:26,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Frame.h:30,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:28,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_stereo.cc:33:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
  162 | EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
      |                                     ^~~~~~~~~~
In file included from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/LoopClosing.h:34,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/LocalMapping.h:25,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:30,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_stereo.cc:33:
/home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:60:80: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
   60 |       typedef Eigen::Map<Matrix<double, Dj, Di>, Matrix<double, Dj, Di>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockTransposedType;
      |                                                                          ~~~~~~^~~~~~~~~~~~
In file included from /usr/local/include/eigen3/Eigen/Core:344,
                 from /home/rakib/Desktop/DroneProject/Codes/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/ImuTypes

richard-elvira, 2020-09-08:
Have you run build.sh file before?

BeardyMan37, 2020-09-08:
Yes, I have. I also ran it on the EuroC dataset and the code works perfectly. It's just when I try to build it for ROS, this problem shows up. I initially thought this problem was due to me having opencv 3.2.0 installed, and libopencv4.2.0 installed, as shown in the warning. I looked around and I found in ROS forums that they are just harmless warninga. I installed opencv 4.2.0 nonetheless. Still no change of result.

kennyjchen, 2020-09-20:
Hi @BeardyMan37, did you ever solve this issue? I am seeing the same problem with an undefined reference to _ZNK2cv8FileNodecviEv. I am trying to compile on Ubuntu 20.04 (which ships with OpenCV 4.2) but have compiled a local copy of OpenCV3.2 and successfully built ORB-SLAM3 via the build.sh script. I see this issue when trying to build the ROS examples via build_ros.sh.

BeardyMan37, 2020-09-20:
No, I still haven’t managed to get it working. I have the same configuration, and I also could build the project, but failied in building ROS. Hoping the developers have seen the issue before.

Ishira1601, 2020-11-16:
Has someone managed to solve this problem. I'm having a similar problem. I guess it's because ros noetic is targeted for python3 and ORB_SLAM works with python2.7. I'm wondering if there's some workaround this problem

MarcusMcKenzie, 2021-01-27:
I believe the issue is that the program cannot find libopencv_imgproc and libopencv_core. I was able to get it to build by copying those files to ORBSLAM/lib and then by modifying the ORB_SLAM3/Examples/ROS/ORB_SLAM3/CMakeLists.txt like this:

`sudo cp /usr/lib/x86_64-linux-gnu/libopencv_core.so /home/marcus/catkin_ws/src/ORB_SLAM3/lib
 
sudo cp /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2.0 /home/marcus/catkin_ws/src/ORB_SLAM3/lib
 
sudo cp /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so /home/marcus/catkin_ws/src/ORB_SLAM3/lib
 
sudo cp /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2.0 /home/marcus/catkin_ws/src/ORB_SLAM3/lib`

`set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
#${Boost_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM3.so

${PROJECT_SOURCE_DIR}/../../../lib/libopencv_core.so
${PROJECT_SOURCE_DIR}/../../../lib/libopencv_imgproc.so
)
`

aprotyas, 2021-04-12:
@MarcusMcKenzie I suspect this makeshift solution might come back to bite if there's any API changes between OpenCV 3.x and OpenCV 4.2 and ORBSLAM2 happens to call incompatible functions 😢 But I can confirm your fix works for me too

Olofforsberg, 2021-07-02:
Hi also stumbled on this issue trying to compile it for ros. I have both opencv 3.2 and 4.5 versions installed in ubuntu 20.04. The fix that worked for me was to change line 33 in CmakeLists.txt in Examples/ROS/ORB_SLAM3/
from:
`find_package(OpenCV 3.0 QUIET)`
to:
`find_package(OpenCV 4.0 QUIET)`

aniket-gupta1, 2022-04-05:
@Olofforsberg Thanks that worked

jragni, 2025-03-01:
in case this doesn't work, [https://github.com/Mechazo11/ros2_orb_slam3](https://github.com/Mechazo11/ros2_orb_slam3) works for me