UZ-SLAMLab/ORB_SLAM3 #98: AssignFeaturesToGrid

qiaozhijian, 2020-09-08:
when it comes to stereo,  should AssignFeaturesToGrid be put on the last one line in Frame constructor?

qiaozhijian, 2020-09-08:
`const cv::KeyPoint &kp = (Nleft == -1) ? mvKeysUn[i]
                                                 : (i < Nleft) ? mvKeys[i]
                                                                 : mvKeysRight[i - Nleft];`
but the default of Nleft is 0.

jjgr3496, 2020-10-20:
Could you please post the exact lines where you think there is a mistake?

The stereo fisheye frame constructor looks fine but you are right that for the stereo pinhole constructor, assignFeaturesToGrid should be called in the end as Nleft is unitialized and that could lead to undesired behavior. We will fix this in the upcomming release. Thank you for your helpful comment!

qiaozhijian, 2020-10-20:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/ef9784101fbd28506b52f233315541ef8ba7af57/src/Frame.cc#L175

function "AssignFeaturesToGrid" uses "Nleft", but  "Nleft" is initialized behind.

jjgr3496, 2020-10-23:
It is indeed a mistake. We will solve it in the next release!
Thank you for your helpful comments.

closed 2020-10-23 by jjgr3496