UZ-SLAMLab/ORB_SLAM3 #102: For fisheye camera, which model is better ?

XinnWang, 2020-09-09:
Hi, thanks for your great work !

i am using a fisheye camera to do monocular slam. After calibration by opencv, there are two models i can use:

1. Using KannalaBrandt8 model.
2. Using cv::fisheye::undistortImage to get undistorted image (it is the central part of fisheye image), then using PinHole model.

i tried both ways to construct small maps in my office, my code is based on 'mono_kitti.cc', and my config file is based on 'TUM_512.yaml'. i only made minor changes to fit my own dataset. i observed different performance:

1. When using KannalaBrandt8 model, orb feature points scatter over image, the resulted point cloud map looks quite loose, contains many noises. (i tried to increase ORBextractor.nFeatures, but bring more noises)
2. When using PinHole model, orb feature points concentrate on edges or corners, the resulted point cloud map looks compact, contains much much fewer noises, furniture can be easily distinguished.

i am confused, it looks like PinHole model is better, but i don't have ground truth so i can not compare them fairly. i am thinking, maybe,  PinHole model is more precise for easy dataset, and KannalaBrandt8 model is more robust for challenging dataset. i want to know have you ever compared the performance of KannalaBrandt8 model and PinHole model for different kind of fisheye dataset ? is there any results to proof which model is better ? Thanks.

jjgr3496, 2020-09-09:
First of all, "visually appealing" a map is is not a good metric for several reasons. First, it is a completely subjective measure and second because you can't really have a standard metric (datasets don't usually provide any map information besides camera poses for computing a ground truth). This is specially critical in monocular vision and sparse maps as the map scale is not fixed. For instance, a map with a smaller estimated scale will have all features closer which will render better contours and structure while for map with a larger estimated scale all points will spread along all the space, making difficult for the human eye to reconstruct shapes.

Being said that, your question is kinda tricky and requires a long explanation. It is important to **keep in mind that this is more about robustness than accuracy**. As you have already noted and experimented with, there are 2 approaches to deal with fisheye cameras in V-SLAM:

1. Use the raw and distorted image (the fisheye images)
2. Undistort images so they follow a pinhole model (the undistorted images)

Approach 2 raises some questions and problems. First of all, you need to define a new calibration matrix for the undistorted images. This lead to 2 scenarios depending of that new calibration:

- Choosing small focal lengths leads to undistorted images with low FoV
- On the other hand, choosing large focal lengths leads to a nice FoV but the central region of the image gets reduced while the outer parts get strongly magnified.

Also, in both cases, you have to interpolate some pixels (i.e. you make them up) and impose an extra (and costly) extra step inside the SLAM pipeline. In any case, this have 2 main drawbacks:

- The magnification effect has a negative effect as objects in image change their size as they move along the image. Also, if you take a look of a fisheye image and its undistorted version, you will notice that angles are not preserved in the undistorted one. All this together can lead to some **difficulties performing descriptor matching**.
- In general, a distorted image will always have a smaller FoV than its counterpart (think for example of a fisheye camera with FoV grater than 180º). This have a really negative impact on the robustness as fisheye images are less prone to get completely occluded due to its FoV (more robust).

Also, approach 1 has the advantage of running with original fisheye images, so no undistortion step is required (this is especially important for hard-constrained real time systems). 

From our experiments (an indoor sequence), we got the following results (RMSE trajectory): 

- If camera motion is smooth, fisheye and undistoted image perform similarly.
- If camera motion is aggressive with sudden movements and turns, fisheye performs significantly better than undistorted images. Undistorted images with short focal lengths (bigger FoV) could still process the sequence but got worse RMSE and undistorted images with long focal lengths (smaller FoV) couldn't even process it as they lost the camera track.

XinnWang, 2020-09-13:
Thanks so much for such a detailed answer !

closed 2020-09-14 by jjgr3496