UZ-SLAMLab/ORB_SLAM3 #104: Build error with ORB_SLAM3

ffrivera0, 2020-09-09:
Hi,

I got a build error when building the ORB3-SLAM and I have no idea about it, could you help me to figure it out? Thanks
I post the error below:
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5196950/log.txt)

choffmann, 2020-09-09:
Have you considered removing the `const` for the allocator?

From
```c++
    typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*>,
        Eigen::aligned_allocator<std::pair<const KeyFrame*, g2o::Sim3> > > KeyFrameAndPose;
```
to
```c++
   typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*>,
        Eigen::aligned_allocator<std::pair<KeyFrame*, g2o::Sim3> > > KeyFrameAndPose;
```
in `include/LoopClosing.h`.
I didn't get this error. It seems your compiler is more pedantic in this regards than mine. Which compiler do you use?

ffrivera0, 2020-09-13:
Thanks. Now I can run it successfully.

closed 2020-09-13 by ffrivera0