UZ-SLAMLab/ORB_SLAM3 #108: ORB_SLAM3 Evaluation

andriskuciks, 2020-09-11:
Hello everyone! 
I have a question about ORB_SLAM3 evaluation. I was wondering whats the things I have to do and what tools I should use to successfully evaluate and get RMSE values. As I looked into ORB_SLAM2 and 3 it seems like there is a python script `evaluate_ate_scale.py` which does the comparison between calculated data and ground truth. But there are no instructions how its done. The ground truth tho is in KITTI dataset format but generated `KeyFrameTrajectory_TUM_Format.txt` is in TUM format. 
So have anyone successfully evaluated ORB_SLAM2/3 results using this script with your own dataset? And what steps I should take or what other tools I should use to get to this result? It will be critical to evaluate the results as we're preparing our own `.bag` dataset to test some of the SLAM algorithms for my master thesis. Any help much appreciated!

closed 2020-09-14 by andriskuciks

KaninchenM, 2020-11-24:
Could you please tell me which evaluation tools do you use at last? Thank you~

andriskuciks, 2020-11-24:
Hi @KaninchenM! 
I settled on this evaluation tool: https://github.com/MichaelGrupp/evo which is straight forward to use. But I ended up writing my own python scripts to do the data processing. The results were similar to the `evo` tool mentioned before

KaninchenM, 2020-11-25:
> Hi @KaninchenM!
> I settled on this evaluation tool: https://github.com/MichaelGrupp/evo which is straight forward to use. But I ended up writing my own python scripts to do the data processing. The results were similar to the `evo` tool mentioned before

I have used evo, too. But after running orbslam3 10 times on Euroc, I got the median rmse(of ATE) =0.0409m( in MH01),  median rmse(of ATE) =0.0459m( in MH02),  median rmse(of ATE) =0.0467m( in MH02), median rmse(of ATE) =0.0360m( in MH03), while the paper's result is 0.025, 0.022, 0.027m respectively. How is your result?
[image]

andriskuciks, 2020-11-25:
> I have used evo, too. But after running orbslam3 10 times on Euroc, I got the median rmse(of ATE) =0.0409m( in MH01), median rmse(of ATE) =0.0459m( in MH02), median rmse(of ATE) =0.0467m( in MH02), [...]

I ran in for couple of times few months ago but didn't record any results. My field of interest is to compare out of the box tracking modules like Intel T265 tracking results and open source algorithm results on our test recording so I didn't really compare the results of running ORBSLAM3 with Euroc dataset and the results presented in the paper.

KaninchenM, 2020-11-26:
> I ran in for couple of times few months ago but didn't record any results. My field of interest is to compare out of the box tracking modules like Intel T265 tracking results and open source [...]

OK, still thank you for your reply!

andriskuciks, 2022-03-31:
Hello @Mechazo11 ,
I haven't done anymore evaluations since this post was created. Unfortunately I do not remember how it's done since it was 1.5 years ago.

jongwonjlee, 2023-06-05:
@Mechazo11  Have you ended up getting the evaluation over the EuRoC dataset to match the results on the ORB SLAM3 paper? I'm recalling this thread as I just realized that ORB-SLAM3 code seems to produce the absolute trajectory errors comparable to its paper only with the files in [`evaluation/Ground_truth/`](https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/master/evaluation/Ground_truth) as ground truth, not the 'original' ones in the [EuRoC dataset](http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/) (what you are referring to as `data.csv`).

For example, when I run ORB-SLAM3 as a stereo-inertial mode over the V1-01 sequence in the EuRoC dataset, it shows 0.026364 m ATE with respect to [`evaluation/Ground_truth/EuRoC_left_cam/V101_GT.txt`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/evaluation/Ground_truth/EuRoC_left_cam/V101_GT.txt) as ground truth, but shows 0.084402 m ATE with respect to the authentic ground-truth provided in the [EuRoC dataset](http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/). Only the former seems comparable to the figure reported in the paper (0.038 m ATE), and I'm wondering what magic has happened to the files in [`evaluation/Ground_truth/`](https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/master/evaluation/Ground_truth) provided as part of this repo!

KaninchenM, 2023-06-05:
您的邮件已收到，我会尽快回复您！

Mechazo11, 2023-06-07:
@jongwonjlee Hi, sorry for the late response. No, my research thrust ended up in a different direction that did not need for me to evaluate MH01 dataset.

Mechazo11, 2023-09-05:
> @Mechazo11 Have you ended up getting the evaluation over the EuRoC dataset to match the results on the ORB SLAM3 paper? I'm recalling this thread as I just realized that ORB-SLAM3 code seems to [...]

@jongwonjlee Hi, apologies for the late reply, I got side tracked in the months of June thru August and could not pick back up on this problem. Yes, I am not really sure what processing was done but with the .csv files the results are not as small as the authors had reported. As of now, I will be sticking to the .csv data files for evaluating the ATE values since using the .csv appears to be producing results similar to what @KaninchenM had reported

MHarbi, 2024-02-26:
For those facing challenges in outcome evaluation, the script [evaluation/evaluate_ate_scale.py](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/evaluation/evaluate_ate_scale.py) and the [evo_ape](https://github.com/MichaelGrupp/evo/wiki/Metrics) metric from the `evo` tool are both effective solutions, with `evo` offering additional benefits. The evaluation of the MH01 trajectory using Stereo setting with `evo_ape` is presented below.
```ssh
$ evo_ape tum ${SLAM3_src}/evaluation/Ground_truth/EuRoC_left_cam/MH01_GT2.txt f_dataset-MH01_stereo_TUM.txt --verbose --align --correct_scale

--------------------------------------------------------------------------------
Loaded 3638 stamps and poses from: *****/ORB_SLAM3/evaluation/Ground_truth/EuRoC_left_cam/MH01_GT2.txt
Loaded 3682 stamps and poses from: f_dataset-MH01_stereo_TUM.txt
--------------------------------------------------------------------------------
Number of poses in reference was reduced from 3638 to 3638.
Number of poses in estimate was reduced from 3682 to 3682.
--------------------------------------------------------------------------------
Synchronizing trajectories...
Found 3638 of max. 3638 possible matching timestamps between...
        *****/ORB_SLAM3/evaluation/Ground_truth/EuRoC_left_cam/MH01_GT2.txt
and:    f_dataset-MH01_stereo_TUM.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning using Umeyama's method... (with scale correction)
Rotation of alignment:
[[ 0.33169362  0.27849104 -0.9013446 ]
 [ 0.94268682 -0.13465278  0.30530345]
 [-0.03634429 -0.95095288 -0.3071933 ]]
Translation of alignment:
[ 4.66955307 -1.83576646  0.93393552]
Scale correction: 1.0072292794463467
--------------------------------------------------------------------------------
Compared 3638 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(with Sim(3) Umeyama alignment)

       max      0.083871
      mean      0.015153
    median      0.012525
       min      0.000670
      rmse      0.018032
       sse      1.182943
       std      0.009775
```

The same outcomes were obtained using the `evaluate_ate_scale.py` script.

```
$ python ${SLAM3_src}/evaluation/evaluate_ate_scale.py ${SLAM3_src}/evaluation/Ground_truth/EuRoC_left_cam/MH01_GT2.txt f_dataset-MH01_stereo.txt --scale 1.0072292794463467 --verbose --verbose2

compared_pose_pairs 3638 pairs
absolute_translational_error.rmse 0.018032 m
absolute_translational_error.mean 0.015153 m
absolute_translational_error.median 0.012525 m
absolute_translational_error.std 0.009775 m
absolute_translational_error.min 0.000670 m
absolute_translational_error.max 0.083871 m
max idx: 1426
compared_pose_pairs 3638 pairs
absolute_translational_error.rmse 0.018032 m
absolute_translational_errorGT.rmse 0.018032 m
```
NOTE: the scale correction was calculated using the `ape` tool.

KaninchenM, 2024-02-26:
您的邮件已收到，我会尽快回复您！

smallQi25, 2024-12-29:
有人使用rgbd_tum测试过orbslam3

KaninchenM, 2024-12-29:
您的邮件已收到，我会尽快回复您！