UZ-SLAMLab/ORB_SLAM3 #109: Disabling the loop closing module

karnikram, 2020-09-12:
Hi, I'm trying to run ORB-SLAM3 without loop correction for a comparison against pure VIO methods. Simply disabling the `mptLoopClosing` thread doesn't work as expected, as it is tightly coupled with the local mapping and tracking threads. The methods discussed [here](https://github.com/raulmur/ORB_SLAM2/issues/256) for ORB-SLAM2 don't work either as the loop detection and closing modules have changed significantly for ORB-SLAM3.

For now I'm commenting out [line1](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/5e591894ec4e23045a8621df7da95cdf659df1b2/src/LoopClosing.cc#L324) and [line2](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/5e591894ec4e23045a8621df7da95cdf659df1b2/src/LoopClosing.cc#L476), and setting `mbLoopDetected` to `false` at both places. This seems to disable the loop correction as seen from the trajectory and map outputs and stdout, but many loop detection steps are still running.

Is there a better way of doing this? It'd be great if this option is built into the code.

jdtardos, 2020-10-19:
ORB-SLAM3 is a SLAM system, not a VIO system, and this influenced all design decisions. In our opinion, switching off loop closing is unfair, as you cannot undo all the related design decisions. It would be like taking out two wheels of a car to compare it with a motorbike.

closed 2020-10-19 by jdtardos

hobbeshunter, 2020-10-19:
Have a look at https://github.com/hobbeshunter/ORB_SLAM3/commit/b7f00c3d9d6bb4c4f8c1cd4954375126b7927210

I made it a parameter in the settings file. 

IMHO, for research purposes it is still very interesting to compare performance with and without loop closure. Especially if you have datasets without ground truth and the only measure you have is if the vehicle returned to the starting point.

Shane922, 2020-12-09:
> Have a look at [hobbeshunter@b7f00c3](https://github.com/hobbeshunter/ORB_SLAM3/commit/b7f00c3d9d6bb4c4f8c1cd4954375126b7927210)
> 
> I made it a parameter in the settings file.
> 
> IMHO, for research purposes it is still very interesting to compare performance with and without loop closure. Especially if you have datasets without ground truth and the only measure you have is if the vehicle returned to the starting point.

Hi, I've tried your no_loop_closure branch and set the parameter -- System.LoopClosing: 0 and 1 , in ORB_SLAM3/Examples/Stereo-Inertial/EuRoC.yaml, and the terminal also return the info that "Disabled Loop Closing ". However, when I compare the positioning precision using evo_ape, I found little difference between the two options that the parameter is set to be 0 or 1. I'm not sure if I missed anything or made any mistakes, and can you give me some simple instructions if it's convenient for you, thanks very much!

hobbeshunter, 2020-12-10:
Hi Shane922,

it really depends on the sequence if loop closing makes a difference or not. Usually it just happens in sequences where the camera goes exploring and then after a long time returns to a previously visited place. Since ORB-SLAM maintains the essential graph it already does a lot of mini loop closing (midterm data association), so very often (e.g. in the TUM-VI room sequences) no real loop closing is necessary.

If it closes a loop it prints a lot of stuff in the terminal (e.g. "PR: Loop detected with Reffine Sim3"). If this does not happen in your sequence then -off course- disabling the loop closing will have no effect.

Shane922, 2020-12-21:
> Hi Shane922,
> 
> if loop closing makes a difference or not depends on the sequence . Usually it just happens in sequences where the camera goes exploring and then after a long time returns to a previously visited place. Since ORB-SLAM maintains the essential graph it already does a lot of mini loop closings (midterm data association), so very often (e.g. in the TUM-VI room sequences) no real loop closing is necessary.
> 
> If it closes a loop it prints a lot of stuff in the terminal (e.g. "PR: Loop detected with Reffine Sim3"). If this does not happen in your sequence then -off course- disabling the loop closing will have no effect.

Thanks very much, it helps a lot!

surfii3z, 2021-03-17:
> Have a look at [hobbeshunter@b7f00c3](https://github.com/hobbeshunter/ORB_SLAM3/commit/b7f00c3d9d6bb4c4f8c1cd4954375126b7927210)
>
> I made it a parameter in the settings file.
>
> IMHO, for research [...]

Thank you for your works. I have adopted it in my experiment and it works nicely.

However, there is a side effect when you set `System.LoopClosing: 0` though. It also disable the re-localization after losing the track.

hobbeshunter, 2021-05-08:
> It also disable the re-localization after losing the track. Does anyone experience the same issue?

Well since re-localization and loop closing are essentially the same, this is a logic consequence.