UZ-SLAMLab/ORB_SLAM3 #110: Segmentation Fault while running ORB-SLAM3.

vislero, 2020-09-14:
Hello,

Thanks for your great working first.

I tested ORB-SLAM3 with fisheye camera (T-265) in live version.
When I tested the ORB-SLAM3 with this sensor, it seems to work fine under normal circumstances. (below figure)

[image]
[image]

By the way, when I tried to run ORB-SLAM3 in the long run, I always got a segmentation error.

So I set up GDB to check for error lines.

And the result we checked is that the KeyFrameCulling() in LocalMapping.cc throws an error.(below figure)
[image]

But I don't know the specific reason for the segmentation error.

What is the cause of this problem?

leo-wkd, 2020-10-29:
Hi, could you please upload your calibration files of T265? I have calibred it using Kalibr but there are still serious drifting problems when I try to run Stereo_Inertial rosnode.

ynma-hanvo, 2022-03-31:
@vislero   ( By the way, when I tried to run ORB-SLAM3 in the long run, I always got a segmentation error.

So I set up GDB to check for error lines )

hi, i am experience the same issue with my mynteye cameras.   
i tested the Stereo / Stereo-Inertial Mode only,  it always crashed with Segmentation error after ruining for some seconds; 

do you have a clue