UZ-SLAMLab/ORB_SLAM3 #111: Output when using rosrun

BritskNguyen, 2020-09-14:
Dear Authors,

Thank you for sharing your work.

I have been able to run the ros examples. However there is no topic published or txt file saved to the folder. Could you please explain how to retreive the estimation result when running as ros node?

BritskNguyen, 2020-09-14:
Ok so to save the trajectory, I add this after ros::spin() in ROS/ORB_SLAM3/ros_stereo_inertial.cc file

```
SLAM.Shutdown();
SLAM.SaveTrajectoryEuRoC("/home/USER/log.txt");

```

So when not node exits, the trajectory is saved to the /home/USER/log.txt file.

closed 2020-09-14 by BritskNguyen

ihamieh, 2020-09-15:
@BritskNguyen this is exactly what i was looking for and I posted this seperatly! I will try this tonight and see what other topics i can output as I'm running ORB_SLAM3 in ROS... I'll keep you posted!