UZ-SLAMLab/ORB_SLAM3 #112: Optimization crash

Norfo, 2020-09-14:
Good day! Thanks for your work. I am using slam on Windows 10. Stereo mode works very well. However, when optimization starting, my application crashes with an error. This happens at about 10-12 keyframes when the optimizer is called. The call stack looks like this.

 	>TestOrbSlam3.exe!g2o::BlockSolver<struct g2o::BlockSolverTraits<-1,-1> >::buildSystem(void)	C++
 	>TestOrbSlam3.exe!g2o::OptimizationAlgorithmLevenberg::solve(int,bool)	C++
 	>TestOrbSlam3.exe!g2o::SparseOptimizer::optimize(int,bool)	C++
 	>TestOrbSlam3.exe!ORB_SLAM3::Optimizer::FullInertialBA(class ORB_SLAM3::Map *,int,bool,unsigned long,bool *,bool,float,float,class Eigen::Matrix<double,-1,1,0,-1,1> *,bool *)	C++
 	>TestOrbSlam3.exe!ORB_SLAM3::LocalMapping::InitializeIMU(float,float,bool)	C++
 	>TestOrbSlam3.exe!ORB_SLAM3::LocalMapping::Run(void)	C++
 	>TestOrbSlam3.exe!std::_LaunchPad<class std::unique_ptr<class std::tuple<void ( ORB_SLAM3::LocalMapping::*)(void),class ORB_SLAM3::LocalMapping *>,struct std::default_delete<class std::tuple<void ( ORB_SLAM3::LocalMapping::*)(void),class ORB_SLAM3::LocalMapping *> > > >::_Go(void)	C++
	>TestOrbSlam3.exe!std::_Pad::_Call_func(void * _Data)Line 209	C++
 	>TestOrbSlam3.exe!thread_start<unsigned int (__cdecl*)(void *),1>(void * const parameter)Line 97	C++

Can you tell me what could be the cause of this problem? I will be glad for any help and will provide any additional information if necessary. Thank you in advance.

electech6, 2020-10-27:
Hi, i have the same problem, have you solved this problem?

Norfo, 2020-10-27:
Unfortunately, not. I am waiting for the new version v0.4 to be released in the hope that there will be changes that will solve this problem.