UZ-SLAMLab/ORB_SLAM3 #117: Issue running ORB_SLAM3 with seperate data

ihamieh, 2020-09-15:
hey guys, i am not seeing any trajectory when I run my own data with ORB_SLAM3. The following is what i get: 
[image]

7H:~/ORB_SLAM3$ rosrun ORB_SLAM3 Mono Vocabulary/ORBvoc.txt Examples/ROS/ORB_SLAM3/TUM_512_IH.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
K1 = 0.00348239
K2 = 0.000715035
K3 = -0.00205324
K4 = 0.000202937

Camera Parameters: 
- fx: 190.978
- fy: 190.973
- cx: 254.932
- cy: 256.897
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: -0.00205324
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
First KF:0; Map init KF:0
New Map created with 53 points
Point distribution in KeyFrame: left-> 53 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 118
mnInitialFrameId = 0
1 Frames set to lost
First KF:2; Map init KF:0
New Map created with 71 points
Point distribution in KeyFrame: left-> 71 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 139
mnInitialFrameId = 119
2 Frames set to lost
First KF:4; Map init KF:2
New Map created with 86 points
Point distribution in KeyFrame: left-> 86 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 433
mnInitialFrameId = 140
3 Frames set to lost
First KF:6; Map init KF:4
New Map created with 65 points
Point distribution in KeyFrame: left-> 65 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 521
mnInitialFrameId = 434
4 Frames set to lost
First KF:8; Map init KF:6
New Map created with 73 points
Point distribution in KeyFrame: left-> 73 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 582
mnInitialFrameId = 522
5 Frames set to lost
First KF:10; Map init KF:8
New Map created with 93 points
Point distribution in KeyFrame: left-> 93 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 614
mnInitialFrameId = 583
6 Frames set to lost
First KF:12; Map init KF:10
New Map created with 67 points
Point distribution in KeyFrame: left-> 67 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 617
mnInitialFrameId = 615
7 Frames set to lost
First KF:14; Map init KF:12
New Map created with 59 points
Point distribution in KeyFrame: left-> 59 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1052
mnInitialFrameId = 618
8 Frames set to lost
First KF:16; Map init KF:14
New Map created with 60 points
Point distribution in KeyFrame: left-> 60 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1107
mnInitialFrameId = 1053
9 Frames set to lost
First KF:18; Map init KF:16
New Map created with 56 points
Point distribution in KeyFrame: left-> 56 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1352
mnInitialFrameId = 1108
10 Frames set to lost
First KF:20; Map init KF:18
New Map created with 86 points
Point distribution in KeyFrame: left-> 86 --- right-> 0
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1463
mnInitialFrameId = 1353
11 Frames set to lost

Anyone knows what's going on?

repa1030, 2022-02-04:
Got same problem with ORB SLAM 3 v1.0. Do you have any solution for this? I already checked the IMU-Camera Calibration multiple times but it is definitively correct.

Maithilishetty, 2022-03-21:
@repa1030 @ihamieh were you able to fix this?