UZ-SLAMLab/ORB_SLAM3 #118: Makefile:83 recipe for target 'all' failed when running ./build.sh

kwads123, 2020-09-16:
I am having a problem running ./build.sh. I have added my full output from the terminal window below. Before this, I was receiving a separate error about find_package not knowing the version of Eigen3 and not accepting it. I can confirm I have installed Eigen3 (via sudo apt-get install libeigen3-dev) and the version is 3.2.92. I was able to bypass this problem by removing the version in find_package in both ORB_SLAM3/CMakeLists.txt and ORB_SLAM3/Thirdparty/g2o/CMakeLists.txt E.g.

find_package(Eigen3 3.1.0 REQUIRED) >>>> find_package(Eigen3 REQUIRED)

Build Error:
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tim/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tim/ORB_SLAM3/Thirdparty/g2o/build
Scanning dependencies of target g2o
[  6%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[  6%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[  9%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
[ 12%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel.cpp.o
[ 16%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/timeutil.cpp.o
[ 19%] Building CXX object CMakeFiles/g2o.dir/g2o/core/solver.cpp.o
[ 22%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_graph_action.cpp.o
[ 25%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel_impl.cpp.o
[ 29%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel_factory.cpp.o
[ 32%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_dijkstra.cpp.o
[ 35%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_factory.cpp.o
[ 38%] Building CXX object CMakeFiles/g2o.dir/g2o/core/batch_stats.cpp.o
[ 45%] Building CXX object CMakeFiles/g2o.dir/g2o/core/marginal_covariance_cholesky.cpp.o
[ 48%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm.cpp.o
[ 51%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o
[ 87%] Building CXX object CMakeFiles/g2o.dir/g2o/core/parameter.cpp.o
[ 58%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_graph.cpp.o
[ 61%] Building C object CMakeFiles/g2o.dir/g2o/stuff/os_specific.c.o
[ 64%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_levenberg.cpp.o
[ 67%] Building CXX object CMakeFiles/g2o.dir/g2o/core/parameter_container.cpp.o
[ 70%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_gauss_newton.cpp.o
[ 70%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.o
[ 70%] Building CXX object CMakeFiles/g2o.dir/g2o/core/matrix_structure.cpp.o
[ 74%] Building CXX object CMakeFiles/g2o.dir/g2o/core/jacobian_workspace.cpp.o
[ 77%] Building CXX object CMakeFiles/g2o.dir/g2o/core/sparse_optimizer.cpp.o
[ 80%] Building CXX object CMakeFiles/g2o.dir/g2o/core/estimate_propagator.cpp.o
[ 83%] Building CXX object CMakeFiles/g2o.dir/g2o/core/cache.cpp.o
[ 90%] Building CXX object CMakeFiles/g2o.dir/g2o/core/factory.cpp.o
[ 93%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
[ 96%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimizable_graph.cpp.o
[100%] Linking CXX shared library ../lib/libg2o.so
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
OPENCV VERSION:
3.3.1
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tim/ORB_SLAM3/build
Scanning dependencies of target g2o
Scanning dependencies of target ORB_SLAM3
[  1%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
[  2%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[  3%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[  4%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/hyper_graph_action.cpp.o
[  6%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/hyper_graph.cpp.o
[... 1126 lines cut ...]
/home/tim/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/hyper_graph.h:148:118: warning: ‘VA2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           void setVertex(size_t i, Vertex* v) { assert(i < _vertices.size() && "index out of bounds"); _vertices[i]=v;}
                                                                                                                      ^
/home/tim/ORB_SLAM3/src/Optimizer.cc:526:42: note: ‘VA2’ was declared here
                 g2o::HyperGraph::Vertex* VA2;
                                          ^
In file included from /home/tim/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:38:0,
                 from /home/tim/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/tim/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/tim/ORB_SLAM3/include/LoopClosing.h:34,
                 from /home/tim/ORB_SLAM3/include/Optimizer.h:26,
                 from /home/tim/ORB_SLAM3/src/Optimizer.cc:20:
/home/tim/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/hyper_graph.h:148:118: warning: ‘VG2’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           void setVertex(size_t i, Vertex* v) { assert(i < _vertices.size() && "index out of bounds"); _vertices[i]=v;}
                                                                                                                      ^
/home/tim/ORB_SLAM3/src/Optimizer.cc:525:42: note: ‘VG2’ was declared here
                 g2o::HyperGraph::Vertex* VG2;
                                          ^
/home/tim/ORB_SLAM3/src/Optimizer.cc:490:53: warning: ‘pIncKF’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         VertexAccBias* VA = new VertexAccBias(pIncKF);
                                                     ^
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

WangYixuan12, 2020-10-17:
Hi! I encounter a similar problem! Have you ever solved it?

kwads123, 2020-10-19:
Unfortunately I haven’t. I have attempted installation on 3 systems now, all running the same version of linux. It has only been successful with 1 and the other two have the error above.

kwads123, 2020-11-13:
I have now found a solution to this issue. I have attempted building ORB SLAM3 on 4 different platforms: A powerful laptop running Ubuntu 16.04 and ROS Kinetic, a laptop running Ubuntu 16.04 and ROS Kinetic in a virtual box and two versions of Intel compute stick running 16.04 and 18.04 with Kinetic and Melodic respectively. The only successful install initially was on the more powerful laptop. The other systems failed to install each time. The problem is caused by the build.sh file. The make commands for each part of the build are 'make -j' commands. This automatically uses all processors for the build. If this slows down the system too much, it will start ending processes in order of priority, causing the compiler that ORB SLAM is using to end. To prevent the issue, in the 'build.sh' file change lines 7, 16 and 31 (the make -j lines) to include a specified number of processors to use for the build. I have implemented this for both compute sticks, both running different versions of Ubuntu, by adding 2 to these lines (e.g. make -j2) as they have 4 core CPUs. After this change, the build process runs fine and the build_ros.sh runs fine. No changes need to be made to the build_ros.sh file.

PiStar, 2020-11-17:
> I have now found a solution to this issue. I have attempted building ORB SLAM3 on 4 different platforms: A powerful laptop running Ubuntu 16.04 and ROS Kinetic, a laptop running Ubuntu 16.04 and ROS [...]

Thank you, it works!!

buemi, 2021-03-30:
Hi.
I  failed to perform the build in any platform I tried, even if I changed the -j value depending on the number of available cores. 
What is the minimum RAM requirement to successfully compile the project?

Kanishk598, 2021-06-27:
> Hi.
> I failed to perform the build in any platform I tried, even if I changed the -j value depending on the number of available cores.
> What is the minimum RAM requirement to successfully compile the project?

You can compile the project with any laptop with any RAM (not too old, like before 2005-ish).
The trick here is to use only a single core of your hardware. 
You can do this by changing the `make -j` to `make` in the two files `build.sh` & `build_ros.sh`.
(Make sure that you change all the multiple lines having `make -j` in the `build.sh` file, as it not only needs to compile the project, but also the third-party dependencies.)

buemi, 2021-07-01:
I'm sorry to tell you that I already tried to compile setting the j value to any allowed value and the memory saturation happens.
I used the Dragonboard 410 C equipped with a Qualcomm® Snapdragon (four cores @1.4GHz).

Kanishk598, 2021-07-01:
> I'm sorry to tell you that I already tried to compile setting the j value to any allowed value and the memory saturation happens.
> I used the Dragonboard 410 C equipped with a Qualcomm® Snapdragon (four cores @1.4GHz).

Remove the J altogether. Replace `make -j n` (n = some number) with just `make`.

buemi, 2021-07-01:
No good news: it doesn't work. I have successfully compiled the third party codes, but I still get (after a few hours) the fatal c++ error "killed signal terminated program cclplus", which is usually caused by running out of memory.

Kanishk598, 2021-07-01:
> No good news: it doesn't work. I have successfully compiled the third party codes, but I still get (after a few hours) the fatal c++ error "killed signal terminated program cclplus", which is usually caused by running out of memory.

Just for clarification here, there are three `-j` flags in the `build.sh` file. You are removing all of them, right?

buemi, 2021-07-02:
Yes, I did

buemi, 2021-07-02:
How can I access to the September's version?

Kanishk598, 2021-07-02:
Do this in the ORB-SLAM3 repo that you cloned: `git checkout ef9784101fb` and then do `git checkout -b September_version`

>### Optional Reading
>What these two commands do is: 
>1. The command `git checkout ef97841` brings your git head to the ef9784101fb commit, which is September's commit on this repo. You can see all commits using the command `git log` (this command is optional and doesn't do anything, apart from printing previous commits).
>2. Since your head is now at a commit (September's commit) and not on a branch, the changes you make won't be commit-able. You need to checkout to a new branch from this commit so that you can save your changes and commit them for future. `git checkout -b September_version` does exactly that.

Now you are on a branch that has September's version of ORB-SLAM3. Now start making the required changes to both the build files and ORB-SLAM3 will compile properly.

buemi, 2021-07-06:
Hi,
sorry, I obtain the same fatal c++ error "killed signal terminated program cclplus"
The "Thirdyparty" code compilation succeeded without any problems, the ORB-SLAM3 itself building process fails after several hours.