UZ-SLAMLab/ORB_SLAM3 #119: Issue running ORB_SLAM3 with Mono_Inertia

ihamieh, 2020-09-16:
The following is the error message when I run ORB_SLAM3 with 

7H:~/ORB_SLAM3$ rosrun ORB_SLAM3 Mono_Inertial Vocabulary/ORBvoc.txt Examples/ROS/ORB_SLAM3/TUM_512_IH.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
K1 = 0.00348239
K2 = 0.000715035
K3 = -0.00205324
K4 = 0.000202937

Camera Parameters: 
- fx: 190.978
- fy: 190.973
- cx: 254.932
- cy: 256.897
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: -0.00205324
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
OpenCV Error: Sizes of input arguments do not match (The matrix size does not match to the number of stored elements) in icvReadMat, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/core/src/persistence.cpp, line 5297
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/core/src/persistence.cpp:5297: error: (-209) The matrix size does not match to the number of stored elements in function icvReadMat

Aborted (core dumped)

KHLau96, 2020-09-19:
Sir, I have the same issue with you. Somebody can help?

ihamieh, 2020-09-19:
I fixed this issue with correct camera parameters calibration and alignment!

closed 2020-09-19 by ihamieh

ns15417, 2020-10-12:
> I fixed this issue with correct camera parameters calibration and alignment!

Hi,  how do you decide the alignment between camera and imu? for imu coordinate, should axis-z be upwards? cuz i saw that Gz in code is set to : `const cv::Mat Gz = (cv::Mat_<float>(3,1) << 0,0,-IMU::GRAVITY_VALUE);`