UZ-SLAMLab/ORB_SLAM3 #120: Question about relocalization in tracking.cc

xieqi1, 2020-09-17:
Hi !
  I notice the loop in line 3240, nCandidates is always 0, and the loop will not run. And I find in ORBSLAM2, nCandidates++ when detect a candidate frame via BOW.It seems it will always fail in relocalization, so is it a bug or you just drop the relocalization method?
Thank you !

ankuPRK, 2020-09-18:
I think they have replaced relocalization with submap merging i.e. every new sequence will start with a new sub-map, which will then be aligned and merged with the one of the submaps in the set of maps we already have, given it meets certain criteria.

xieqi1, 2020-09-19:
@ankuPRK Sure, I agree with your opinion.Thank you for your reply~

closed 2020-09-19 by xieqi1

ankuPRK, 2020-09-19:
welcome :)