UZ-SLAMLab/ORB_SLAM3 #122: Orbslam3 RGBD with Azure Kinect

martinakos, 2020-09-17:
I've been trying to run orbslam3 with the azure kinect camera without much success so far. I'll explained what I've tried so far in hope someone can give some help.

I've been using the orbslam3 ROS node and I the [Azure kinect ros node](https://github.com/microsoft/Azure_Kinect_ROS_Driver) from which I have some rosbags that I'm using.

I've tried RGB-D tracking but tracking is very poor. In an attempt to diagnose the problem, I've tried to just track the color image stream and thereby verify the calibration. It tracks reasonably well if I use the undistorted color image stream (provided by the Azure Kinect ros node) and a PinHole camera model without any distortion parameters. But if I use the distorted color image stream and I add the distortion parameters, provided by the Azure Kinect ros node, it doesn't track well. Which makes me suspect that the distortion parameters is where the problem is. 

The Azure Kinect uses a  rational_polynomial which is 8 distortion parameters. In principle it is compatible with the OpenCV as we can see [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61951daac782234f4f28322c0904ba1c4702d0ba/examples/opencv_compatibility/main.cpp#L97).   

The rational_polynomial distortion parameters are available in the /color/camera_info:

```
rostopic echo /color/camera_info

header: 
  seq: 307
  stamp: 
    secs: 1599493078
    nsecs: 628309914
  frame_id: "camera_color_optical_frame"
height: 720
width: 1280
distortion_model: "rational_polynomial"
D: [0.6705218553543091, -2.838226795196533, 0.00039167937939055264, -0.0004037024627905339, 1.6206018924713135, 0.5483986735343933, -2.6671268939971924, 1.5503228902816772]
K: [606.34716796875, 0.0, 638.1181640625, 0.0, 606.2462768554688, 365.56170654296875, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [606.34716796875, 0.0, 638.1181640625, 0.0, 0.0, 606.2462768554688, 365.56170654296875, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

I've tried to remap the above distortion parameters into a PinHole camera in this way:

```
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV)
Camera.fx: 606.34716796875
Camera.fy: 638.1181640625
Camera.cx: 606.2462768554688
Camera.cy: 365.56170654296875

Camera.k1: 0.6705218553543091
Camera.k2: -2.838226795196533
Camera.p1: 0.00039167937939055264
Camera.p2: -0.0004037024627905339
Camera.k3: 1.6206018924713135
Camera.k4: 0.5483986735343933
Camera.k5: -2.6671268939971924
Camera.k6: 1.5503228902816772

Camera.width: 1280
Camera.height: 720
```

Although orbslam3 only recognises k1,k2,p1,p2,k3.  With this distortion parameters tracking is poor in monocular (compared to tracking with no distortion parameters and undistorded image stream), and very poor with RGB-D.

I have tried the KannalaBrandt8 camera model too. And remap the 4 first distortion parameters from camera_info in to the k1,k2,k3,k4 parameters that orbslam3 reads if the KannalaBrandt8 camera model is used. But tracking in monocular is quite poor too.

I guess that if I can't get orbslam3 tracking in monocular with the distortion parameters I get from the Azure Kinect, it's not going to work in RGB-D either. I don't know if it is possible at all to just map the rational_polynomial distortion parameters into the PinHole distortion parameters or the KannalaBrandt8 distortion parameters. Has anybody had any experience with Azure Kinect cameras?

tenneck, 2020-11-13:
Hi,
did you have any success on that issue? It would be very helpful if you can share some experiences.

martinakos, 2020-11-17:
I've been working on something else while waiting for some answer to this issue. So no progress from me.

zhengjialiang2, 2021-01-11:
you can duplicate the codes that reading parameters in "tracking.cc" with k4,k5,k6  you have:
node = fSettings["Camera.k3"];
        if(!node.empty() && node.isReal())
        {
            mDistCoef.resize(5);
            mDistCoef.at<float>(4) = node.real();
        }

        if(b_miss_params)
        {
            return false;
        }

martinakos, 2021-01-18:
@zhengjialiang2 That worked! Thanks!

closed 2021-01-19 by martinakos