UZ-SLAMLab/ORB_SLAM3 #124: [rospack] Error: package 'ORB_SLAM3' not found

flydragon2018, 2020-09-18:
I temporarily comment out  find_package(glog) so that I can finish build_ros.sh

However, when I run the example,package 'ORB_SLAM3' not found.

sudo make install
[ 22%] Built target Mono
[ 55%] Built target MonoAR
[ 77%] Built target RGBD
[100%] Built target Stereo
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/_setup_util.py
-- Up-to-date: /usr/local/env.sh
-- Up-to-date: /usr/local/setup.bash
-- Up-to-date: /usr/local/local_setup.bash
-- Up-to-date: /usr/local/setup.sh
-- Up-to-date: /usr/local/local_setup.sh
-- Up-to-date: /usr/local/setup.zsh
-- Up-to-date: /usr/local/local_setup.zsh
-- Up-to-date: /usr/local/.rosinstall
-- Installing: /usr/local/lib/pkgconfig/orb_slam3.pc
-- Up-to-date: /usr/local/share/orb_slam3/cmake/orb_slam3Config.cmake
-- Up-to-date: /usr/local/share/orb_slam3/cmake/orb_slam3Config-version.cmake
-- Up-to-date: /usr/local/share/orb_slam3/package.xml

gabriel-azeredo, 2022-11-09:
I followed Issue #636 and it worked for me

narnfawn, 2024-07-18:
我遇到了同样的问题，请问您解决了吗？