UZ-SLAMLab/ORB_SLAM3 #125: Config parameters. IMU

dinimar, 2020-09-21:
I have no idea what IMU parameters mean. Could you please explain them and how to calculate it?
Thanks in advance.

### IMU Parameters
IMU.NoiseGyro: 1.7e-04 # 1.6968e-04 
IMU.NoiseAcc: 2.0e-03 # 2.0000e-3
IMU.GyroWalk: 1.9393e-05 
IMU.AccWalk: 3.e-03 # 3.0000e-3

dinimar, 2020-09-21:
I've done imu calibration via https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration. I have the following output:
```
{
    "device_type": "D435i",
    "imus": [
        {
            "accelerometer": {
                "bias": [
                    0.03012660354807157,
                    -0.08302050580743126,
                    0.3764705528882397
                ],
                "scale_and_alignment": [
                    1.0211146183215476,
                    -0.0021680902438522285,
                    0.06452208267420263,
                    0.007454997349000683,
                    1.0136290929807084,
                    0.00035387854221569093,
                    0.0042570229700617385,
                    -0.027620306596414124,
                    1.0136413699483708
                ]
            },
            "gyroscope": {
                "bias": [
                    -0.0025199165869275843,
                    0.0003376978004677162,
                    -0.002784613713486498
                ],
                "scale_and_alignment": [
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0,
                    0.0,
                    0.0,
                    0.0,
                    1.0
                ]
            }
        }
    ]
}
```

weblucas, 2020-09-22:
No, noise is intrinsic of the type of imu. for position you need to calibrate the transform between camera and imu and for the noise you can use higher values for intel imu (cheap imu) and lower for high-quality imus such as from the vi-sensor.
[here](https://github.com/ethz-asl/realsense/tree/devel/d435i)

dinimar, 2020-10-08:
After I'd done my independent research about WhiteNoise and RandomWalk parameters, I found some tools to calculate those parameters.
Tools:
- [rpng/kalibr_allan](https://github.com/rpng/kalibr_allan) - works on MATLAB, requires rosbag of sensor data
- [GAVLab/allan_variance](https://github.com/GAVLab/allan_variance) - works on python, requires rosbag of sensor data

The second repo is more suitalbe solution for me because I don't have MATLAB, but there is another drawback. The repo doesn't have random walk calculation. So, I forked the repo ([my repo](https://github.com/dinimar/allan_variance)) and add missed calculation.

I tested on XSENS Mti-700 G dataset showed in [rpng/kalibr_allan](https://github.com/rpng/kalibr_allan) and 've got equivalient results. Testing on my own dataset I figured out temperature influence on sensor data. You should get flat lines for both acceleration and gyroscope plots as showed below:
[image]
[image]

Otherwise, you will have intervals with outliers. 
To prevent outliers you need:
1. Record dataset that includes imu and temperature data(rosbag)
2. Plot It (rqt_plot, python-script using numpy, matplotlib)
3. Check is there any correlation between temperature and IMU data
4. Crop your dataset to the range where is no oultiers (because in my case IMU sensor was heating up first 30 mins. In first 30 mins there were outliers, after no at all. Graphs above show that cropped dataset)

I calculated parameters using forked repo. I've got expected plot for acceleration and bad plot for gyroscope (because it doesn't seem like allan deviation plot and therefore there wrong random walk calculation). Graphs of gyroscope plot and hypothetical allan deviation plot are showed below. If I'm wrong or you have any solution for this, please email me.
[image]
[image]

closed 2020-10-12 by dinimar