UZ-SLAMLab/ORB_SLAM3 #127: relocazliation wrong!

nlwmode, 2020-09-22:
when I played the mono-imu bag file 2 times , in the start of the second time , 
it output: and why it happened?
`
Creation of new map with id: 1
Exits current map 
Saved map with ID: 0
Creation of new map with last KF id: 306
First KF:306; Map init KF:306
New Map created with 224 points
Point distribution in KeyFrame: left-> 224 --- right-> 0
build optimization graph
Not preintegrated measurement
Segmentation fault (core dumped)
`

wpbird007, 2020-09-29:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/ef9784101fbd28506b52f233315541ef8ba7af57/src/Tracking.cc#L3230
you should after this line ,add  nCandidates++; maybe fix the bug

qingyichen, 2020-10-10:
I got the same problem. Did you figure out the cause?

nlwmode, 2020-10-11:
> I got the same problem. Did y

nlwmode, 2020-10-11:
> I got the same problem. Did you figure out the cause?

The code is so complicated the I can not figure it out , even find the reason. I just think the mono-inertial system is not robust as the video demo.