UZ-SLAMLab/ORB_SLAM3 #128: how to use the SaveAtlas() and LoadAtlas() for relocalization?

nlwmode, 2020-09-22:
I use the origin SaveAtlas() and LoadAtlas() for relocalization, and I also add the function in ShutDown()，and it execte worng!
It is a bit different to ORB_SLAM2  :< 
So how to use the atlas for relocalization?

wpbird007, 2020-09-29:
there is a  bug in  MapPoint::PreSave, detele KeyFrame when iter,you should fix it ,and then fix Tracking::Relocalization bug,and write a new function InitWithMap,and you can use the atlas map

schizzz8, 2020-09-29:
@wpbird007 I'm very interested in this issue.

Please, can you provide more details about the bugs you mention (e.g. source files, line numbers)???

Thanks

thomCastillo, 2020-10-23:
I'm also working on this topic. How did you solve this problem?

colorDogs, 2020-10-26:
@nlwmode can you provide more details about the bugs you mention (e.g. source files, line numbers)???

thomCastillo, 2020-10-26:
Hello @wpbird007 and @colorDogs 
I opened commented lines which about ATLAS in the code ([https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L99](url) and [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L918](url)), but "The program has unexpectedly finished" on the [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/MapPoint.cc#L591](url) line while saving Atlas. Therefore I commented this line and then I could saved Atlas successfully, but when I loaded the Atlas it tries to merge the maps but it gives the "The program has unexpectedly finished" error on the [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Sim3Solver.cc#L488](url) line this time. I tried room1 and corridor1 datasets like [https://www.youtube.com/watch?v=Rw_xP6lDizw](url). Why these lines are commented, is there a bug on the Atlas part of the code? But it seems to work the ORB SLAM3 videos on youtube. Do you know how to use Atlas and merge maps?

colorDogs, 2020-10-26:
> the

Now I only save the map successfully, if you read the map successfully, please contact me

colorDogs, 2020-10-26:
> Hello @wpbird007 and @colorDogs
> I opened commented lines which about ATLAS in the code ([https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L99](url) and [...]

Now I only save the map successfully, if you read the map successfully, please contact me

thomCastillo, 2020-10-27:
> > Hello @wpbird007 and @colorDogs
> > I opened commented lines which about ATLAS in the code ([https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L99](url) and [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L918](url)), but "The program has unexpectedly finished" on the [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/MapPoint.cc#L591](url) line while saving Atlas. Therefore I commented this line and then I could saved Atlas successfully, but when I loaded the Atlas it tries to merge the maps but it gives the "The program has unexpectedly finished" error on the [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Sim3Solver.cc#L488](url) line this time. I tried room1 and corridor1 datasets like [https://www.youtube.com/watch?v=Rw_xP6lDizw](url). Why these lines are commented, is there a bug on the Atlas part of the code? But it seems to work the ORB SLAM3 videos on youtube. Do you know how to use Atlas and merge maps?
> 
> Now I only save the map successfully, if you load the map successfully, please contact me

Hello, @colorDogs 
I added atlas file path at the and of the line https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/mono_inertial_tum_vi.cc#L118. When you add the last parameter the map is loading [ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::IMU_MONOCULAR, true, 0, file_name, atlas_file_path);]. I can see the previous map which I saved on the screen, but while the maps are merging there is an error.

colorDogs, 2020-10-27:
> Hello, @colorDogs
> I added atlas file path at the and of the line https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/mono_inertial_tum_vi.cc#L118. When you add the last [...]

Thank you very much for your reply
I'm ready to try
I'll let you know the results later.

colorDogs, 2020-10-28:
Hello, @wpbird007 
I finished saving and loading the map (but didn't see MapPoint on Pangolin), but a few seconds later a core dump will occur. My guess is that Altas's multi-map mechanism caused the problem. There was a problem with the map merge. I am debugging the cause of what happened. If you have any idea, please contact me

colorDogs, 2020-11-03:
@wpbird007  老哥加个联系方式呗? 还在做slam相关的事情吗?

thomCastillo, 2020-11-03:
@colorDogs I am still working on Atlas.My email adress thom.castillo@yandex.com.I changed some header files but there are some problems.If you contact me , we may discuss these problems.

naoki-17, 2020-11-05:
I have saved and loaded Atlas successfully and found that main problem occurred at the serialization of abstract GeometricCamera class, a self-defined class is a good idea to be serialized and then post-loaded to different camera module class. Also, there are some small problem in postload function to give a value for some pointers such as mpKFlowerID.

colorDogs, 2020-11-05:
> I have saved and loaded Atlas successfully and found that main problem occurred at the serialization of abstract GeometricCamera class, a self-defined class is a good idea to be serialized and then [...]

加个qq联系吧

thomCastillo, 2020-11-05:
@naoki-17 
I commented this line(https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Atlas.h#L62) on Atlas header file and uncommented this line(https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Atlas.h#L63). Thus I solved the geometric camera problem.
I added these on
ar & mPKFinitial;
ar & mpKFlowerID;
 (https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Map.h#L68)

When I run the code it stopped on optimizar cc in this line.
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Optimizer.cc#L7202
It gives the errors below:
(https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Thirdparty/g2o/g2o/core/sparse_optimizer.cpp#L83)
(https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Thirdparty/g2o/g2o/core/sparse_optimizer.cpp#L253)

naoki-17, 2020-11-06:
> @naoki-17
> I commented this line(https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Atlas.h#L62) on Atlas header file and uncommented this [...]

naoki-17, 2020-11-06:
@thomCastillo 
It stopped at LocalBundleAdjustment when I run monocular datasets，which seems the same problem.
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/ef9784101fbd28506b52f233315541ef8ba7af57/src/LoopClosing.cc#L1722
 It failed at 
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/ef9784101fbd28506b52f233315541ef8ba7af57/Thirdparty/g2o/g2o/core/block_solver.hpp#L75
Now I'm debugging. If any progress, shall we share here, thanks
More over, if LocalBundleAdjustment is umcommented, map merge will be finished perhaps with accuracy loss but the work can go on.

353264241, 2020-11-09:
> @thomCastillo
> It stopped at LocalBundleAdjustment when I run monocular datasets，which seems the same problem. [...]

Hello naoki-17 ,I come across the same segment fault in g2o block_solver.hppp L75. And I guess this is because we use the g2o in debug mode and the g2o provided by anthor has the error used in debug.

richard-elvira, 2020-11-09:
We are working in the Load/Save feature and solve the problem of relocalization in the Atlas. The Load/Save feature will be publish at the end of November, we hope to resolve the bug in relocalization in the same realease.

353264241, 2020-11-09:
> We are working in the Load/Save feature and solve the problem of relocalization in the Atlas. The Load/Save feature will be publish at the end of November, we hope to resolve the bug in relocalization in the same realease.

Thanks for sharing this news, suggest using google protobuf to serialize the `KeyFrame` and `MapPoint`,which is a more convenient and debug-friendly serialization library on multi platforms.

naoki-17, 2020-11-12:
> Hello naoki-17 ,I come across the same segment fault in g2o block_solver.hppp L75. And I guess this is because we use the g2o in debug mode and the g2o provided by anthor has the error used in debug.

Hi，I got the same problem both in debug and release mode. Do you mean that it is a idea to use original g2o?

honggesmile, 2020-11-27:
> 加个qq联系吧

老哥，你qq多少，能加个qq请教下吗

honggesmile, 2020-11-30:
> We are working in the Load/Save feature and solve the problem of relocalization in the Atlas. The Load/Save feature will be publish at the end of November, we hope to resolve the bug in relocalization in the same realease.
hi,richard-elvira .thanks for your works,that's so great!.  I also have tried to save and load map,  I have saved Atlas successfully，but when I load map ,the following error occurred：

```
Starting to read the save text file  file name:my_atlas.osa
terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  input stream error
Aborted (core dumped)
```

I only comment    EraseObservation(pKFi) in MapPoint::PreSave(set<KeyFrame*>& spKF,set<MapPoint*>& spMP).

I cannot find any solutions,can you give me some advices.
Looking forward to your reply,thanks very much.

honggesmile, 2020-12-11:
> I have saved and loaded Atlas successfully and found that main problem occurred at the serialization of abstract GeometricCamera class, a self-defined class is a good idea to be serialized and then [...]

Hi，i 

> Hello, @wpbird007
> I finished saving and loading the map (but didn't see MapPoint on Pangolin), but a few seconds later a core dump will occur. My guess is that Altas's multi-map mechanism caused the [...]

老哥，您好，我也是保存并且加载了地图，但是在Pangolin中看不到地图点(I finished saving and loading the map (but didn't see MapPoint on Pangolin),)，请问您解决了吗?

franciscopower, 2021-04-08:
Hi! I've been trying to solve it, but with no success either... Any news on this issue?

neo-knight-td, 2021-04-20:
@richard-elvira any news on that ?

neo-knight-td, 2021-05-04:
Hi all,

Just made the pull request with the Save/Load functionality : https://github.com/UZ-SLAMLab/ORB_SLAM3/pull/310

Hope it helps.

Leonard-Yao, 2021-05-07:
Hi, I try to execute the example  stereo_save and stereo_load from [](https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/dcbb81929fae979384582e9c64c74a30e117cd86), and it worked well. However, I executed load program using the TUM-VI dataset , it crashed with the problem of "Segmentation          fault core dumped ". Whether or not it support the fisheye camera to save and load map? @richard-elvira

neo-knight-td, 2021-05-07:
@Leonard-Yao I did not test with inertial camera. Did you try uncommenting the  `ar & mvpBackupCamKan;` line in Atlas.h ?

honggesmile, 2021-05-07:
> @Leonard-Yao I did not test with fish-eye camera. Did you try uncommenting the `ar & mvpBackupCamKan;` line in Atlas.h ?

I tried to serialize the MvpbackupCamkan, but in IMU mode, the location was still wrong：
`1540 mLastFrame mVw :[]
1545 state:2
1548 state:2
1549 twb1:[2.5284798;
 0.55175859;
 2.6244438]
1550 Vwb1:[-0.18642974;
 0.0013929128;
 0.032918643]
1551 t12:0.04999995231628418
1552 Gz:[0;
 0;
 -9.8100004]
1553 Rwb1:[-0.040686633, 0.99660075, -0.071633935;
 0.33813688, 0.081196956, 0.93758726;
 0.94021684, 0.013925038, -0.3402915]
1554 mLastFrame.mImuBias:-0.026520704850554466
1557 state:2
1560 state:2
1567 state:2
addVertex: FATAL, a vertex with ID 391 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1569 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1570 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1571 has already been registered with this graph
addVertex: FATAL, a vertex with ID 392 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1572 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1573 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1574 has already been registered with this graph
addVertex: FATAL, a vertex with ID 393 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1575 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1576 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1577 has already been registered with this graph
addVertex: FATAL, a vertex with ID 394 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1578 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1579 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1580 has already been registered with this graph
addVertex: FATAL, a vertex with ID 395 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1581 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1582 has already been registered with this graph
addVertex: FATAL, a vertex with ID 1583 has already been registered with this graph
`

can you give me some advices, thanks

neo-knight-td, 2021-05-07:
@honggesmile did you try IMU with pinhole camera ? In EuRoC dataset for example ?

Leonard-Yao, 2021-05-08:
@neo-knight-td Thank you  for  your commentary. I uncommented the "ar & mvpBackupCamKan" and deleted "final" in line 36 from include/CameraModels/KannalaBrandt8.h. I'm not sure it's or not the best way to solve boost lib bug.Now the load program can achieve "*Merged detected". However, it also crashed with the problem of "Segmentation fault,  core dumped " from time to time.

Leonard-Yao, 2021-05-08:
@honggesmile IMU with  camera is not in my schedule at the moment. If  have any solution, I will let you know as soon as possible.

franciscopower, 2021-05-14:
Hi! I executed the save and load functions, but whenever the program ends in a situation where it lost tracking, it cannot save and crashes with the error `segmentation fault (core dumped)` , both in the mono_euroc and stereo_euroc datasets (I only tested with these two).

neo-knight-td, 2021-05-15:
@franciscopower, did you try the example scripts ? Is it that you try to save while tracking is lost or that you have had a tracking lost during the sequence ? On what sequence of the EuRoC dataset do you encounter the problem ?

neo-knight-td, 2021-05-15:
@Leonard-Yao did you also remark that you get a segmentation fault when you have a tracking lost on TUM dataset ?

franciscopower, 2021-05-17:
> @franciscopower, did you try the example scripts ? Is it that you try to save while tracking is lost or that you have had a tracking lost during the sequence ? On what sequence of the EuRoC dataset do you encounter the problem ? Did you try running ORB-SLAM 3 in Debug mode and launching it with gdb ?

I tested the example scripts on the MH01 Euroc dataset. At the end of the set, when the drone is landing, there is a moment when it looses tracking and, sometimes, cannot recover before saving, so it tries to save while tracking is lost. In those occasions I get the segmentation fault error. If I remove those frames from the dataset (cut the sequence short), it saves and loads properly. However, in the monocular mode, as soon as it detects a merge, it crashes with a segmentation fault error. 
I modified the stereo example to both load and save the atlas, and it saves correctly (provided it did't loose localization right before saving) and loads the saved map correctly. However, after loading the map and completing the sequence, it often fails to save again, crashing with the segmentation fault error.

Jiaxin8122, 2021-08-21:
> 加个qq联系吧

Could

> @wpbird007 老哥加个联系方式呗? 还在做slam相关的事情吗?

Hello, 
Could you add me with qq or email?  812230984  Thx!
Regards

closed 2021-12-23 by jdtardos