UZ-SLAMLab/ORB_SLAM3 #135: Can not deal with the real data from RealSenseD435i

gaotao19890725, 2020-09-26:
Dear author:
When I input the following command in the terminal:
rosrun ORB_SLAM3 RGBD /home/gaotao/data/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/gaotao/data/ORB_SLAM3/Examples/RGB-D/TUM1.yaml

nothing shows, it is very strange, can you give me some suggestions:

[image]

schizzz8, 2020-09-30:
You can check two things:

1. make sure that the ROS node is subscribing to the right camera topics
2. make sure that the camera parameters (fx,fy,cx,cy) are the right ones

lbq779660843, 2020-10-12:
I meet the same problem too and I'm using RealSenseD415.

richard-elvira, 2020-10-20:
Notice than ORBSLAM deffine a set of topics where gather the images, in line 110-111 at file "Examples/ROS/ORB_SLAM3/src/ros_stereo.cc":
message_filters::Subscriber<sensor_msgs::Image> left_sub(nh, **"/camera/left/image_raw"**, 1);
message_filters::Subscriber<sensor_msgs::Image> right_sub(nh, **"/camera/right/image_raw"**, 1);

**"/camera/left/image_raw"** identify the rostopic where the camera must send the images. You can change the topics in ORBSLAM or redirect them to those topics with ros commands.

RealSense D435i use these topics to send the images:
/camera/color/image_raw <- Color
/camera/depth/image_rect_raw  <- Depth

lbq779660843, 2020-10-21:
@richard-elvira It works, thank u.

closed 2020-10-21 by richard-elvira