UZ-SLAMLab/ORB_SLAM3 #140: Does ORB_SLAM3 update the camera extrinsic calibration?

ihamieh, 2020-09-29:
I'm not sure if ORB_SLAM3 outputs or updates the camera extrinsic calibration? I was able to output the camera and IMU trajectories but not sure about camera extrinsics? can someone please let me know?

jdtardos, 2020-10-05:
No, you need to provide the camera extrinsic calibration, and ORB-SLAM3 does not modify it.

closed 2020-10-05 by jdtardos

ihamieh, 2020-10-06:
@jdtardos Do you estimate it somehow? Is camea extrinsic a constant input?

jdtardos, 2020-10-19:
Extrinsic calibration is the relative poses between cameras in a stereo system, or the relative pose between the camera(s)  and the IMU sensor. A stereo system can be calibrated with the calibration tools in OpenCV. A visual-inertial system can be calibrated with kalibr.

Chandanpanda, 2023-03-15:
where do you provide the extrinsic parameters?