UZ-SLAMLab/ORB_SLAM3 #145: Assertion (mit->second>=3) failing in LocalInertialBA function

saiprakash-c, 2020-10-09:
Hi,

I am running the monocular-inertial version of ORBSLAM3 with the following configuration:

- Robot: A smart wheelchair 
- Camera: Intel RealSense D435i
- Camera frequency: 15 Hz
- IMU: BMI055(embedded inside the camera) 
- IMU frequency: ~ 200 Hz

The initialization phase happens smoothly. But after initialization, the following assertion fails when the LocalMapping thread calls LocalInertialBA from optimizer.cc. 

_**~/git/ORB_SLAM3/src/Optimizer.cc:4878: static void ORB_SLAM3::Optimizer::LocalInertialBA(ORB_SLAM3::KeyFrame*, bool*, ORB_SLAM3::Map*, bool, bool): Assertion `mit->second>=3' failed.**_

If I remove the assertion, the LocalInertialBA runs with no further problem, but the next keyframe gets added at the wrong location. 

Has anyone run into this issue? May I know what causes this error?

Thanks a lot for your help!

Sai Prakash Reddy C

mihaibujanca, 2020-10-11:
Getting the same error about 20% of the time when running on EUROC-MAV sequences

xzmbb, 2020-11-17:
> Hi,
>
> I am running the monocular-inertial version of ORBSLAM3 with the following configuration:
>
> * Robot: A smart wheelchair
> * Camera: Intel RealSense D435i
> * Camera frequency: 15 Hz
> * IMU: [...]

is solved?  thanks

saiprakash-c, 2020-11-17:
No, I couldn't figure out what triggers this error.

hwb0314, 2020-12-15:
> Hi,
>
> I am running the monocular-inertial version of ORBSLAM3 with the following configuration:
>
> * Robot: A smart wheelchair
> * Camera: Intel RealSense D435i
> * Camera frequency: 15 Hz
> * IMU: [...]

I meet the same problem. Is it solved?

slamhuangxin, 2021-02-01:
I meet the same problem. Is it solved?

guchangjun, 2021-04-15:
> No, I couldn't figure out what triggers this error.

Compile all code with the release. by this way , we avoid this error.

guchangjun, 2021-04-15:
Compile all code with the same release method, e.g., g2o, voc, and orbslam3.

xinruozhishui201314, 2021-04-20:
@guchangjun 不太理解，能详细解释一下吗？

Keeeeeeevin, 2021-07-01:
@xinruozhishui201314 when compiled in debug mode, i encountered the assert(mit->second>=3) problem in localmapping thread. but if i compiled the project in release mode (as well as g2o ThirdParty), the issue disappeared. I yet have not figured out the reason behind, but it works. The only drawback is cant debugging.

robin1987z, 2021-08-30:
comment below code can solve this problem.

#if 0
    //cout << "Total map points: " << lLocalMapPoints.size() << endl;
    for(map<int,int>::iterator mit=mVisEdges.begin(), mend=mVisEdges.end(); mit!=mend; mit++)
    {
        assert(mit->second>=3);
    }
#endif

zhangpeng66, 2022-03-06:
> 

why?who knows the questions ?

MisakaRua, 2022-06-08:
I guess this is because of the implementation of the macro assert. assert will be regard as a fatal error in debug mode but will be regard as nothing in release mode. The implementation in GCC will be like:
`
#ifdef NDEBUG
#define assert(expr) (__ASSERT_VOID_CAST(0))
#else
#define assert(expr) (static_cast<bool>(expr) ? void(0) : __assert_fail(#expr, __FILE__, __LINE__, __ASSERT_FUNCTION))
#endif
`
The answer may be that in release mode the compiler defines the macro NDEBUG, which make the compiler ignore assert() function, and nothing happens when mit->second >= 3. However, the condition will trigger the __assert_fail() function in Debug mode.
In my opinion, this line may not affect the loop closing result. Maybe this line can be removed when using it.

Deep-thinker05, 2022-09-30:
Hi,
I am running the monocular-inertial version of ORBSLAM3 on Euroc-MAV dataset and ecountering this error: 
_**~/git/ORB_SLAM3/src/Optimizer.cc:4878: static void ORB_SLAM3::Optimizer::LocalInertialBA(ORB_SLAM3::KeyFrame*, bool*, ORB_SLAM3::Map*, bool, bool): Assertion `mit->second>=3' failed.**_

Please let me know the intuition behind assertion **mit->second >=3**.
Is removing this assertion correct way of solving the problem or will it give wrong results? I thought of resetting the active map  when this assertion fails but then it is resetting and losing the entire map very often in upcoming frames. is there any other alternative?
Please help.

hongyeah314, 2023-03-18:
I only meet this problem while compiling program with"DEBUG" .when i use Release mode ,this problem is missing but the cam pose was wrong

ZOMAk, 2025-04-30:
src/Optimizer.cc from 2834 line to 2838 line:
// //cout << "Total map points: " << lLocalMapPoints.size() << endl;
// for(map<int,int>::iterator mit=mVisEdges.begin(), mend=mVisEdges.end(); mit!=mend; mit++)
// {
//     assert(mit->second>=3);
// }