UZ-SLAMLab/ORB_SLAM3 #148: stuck on "try to initialize" Mono-IMU

leechangyo, 2020-10-10:
I've spent a whole day trying to solve the "try to initialize" problem using Mono-Imu.

only use Mono, it works fine

but if add IMU, keep failing "Try to initialize".

i use camera attached with IMU sensor, therefore, i set TBC 
to [1, 0,0,0,
     0,1,0,0,
     0,0,1,0
     0,0,0,1]

do you know how to solve this problem? even I use turtlebot of gazebo, it keeps failing to initialize.

ns15417, 2020-10-12:
What do you mean by "Try to initialize"?  the Initialization never succesed? it should be first initialized by mono and then add IMU data

leechangyo, 2020-10-12:
> What do you mean by "Try to initialize"? the Initialization never succesed? it should be first initialized by mono and then add IMU data

yes, the initialization never success by mono and IMU.

cgringmuth, 2020-10-12:
Even if you attached the camera to the IMU there is a translation part and in most cases a small rotation part, as it is almost impossible to align any sensor by hand. So it is always good to calibrate your set-up.

You can use [kalibr](https://github.com/ethz-asl/kalibr) to calibrate your set-up.

leechangyo, 2020-10-12:
> Even if you attached the camera to the IMU there is a translation part and in most cases a small rotation part, as it is almost impossible to align any sensor by hand. So it is always good to calibrate your set-up.
> 
> You can use [kalibr](https://github.com/ethz-asl/kalibr) to calibrate your set-up.

Thank you for your advice. 
we go try this, and leave a comment once we solved or not

kingjin94, 2020-11-27:
@leechangyo Did you manage to get it going with kalibr?