UZ-SLAMLab/ORB_SLAM3 #150: Stereo-Inertial. Have drifting i

dinimar, 2020-10-12:
After I'd done IMU parameters evaluation in #125, I tested new parameters. I got following result ([video](https://yadi.sk/i/29cOLAby4BdK6w))

Now, I'm facing issues:
- I need to constantly move camera to prevent SLAM relocalizing
- I

jdtardos, 2020-10-19:
Repetitive patterns like the calibration templates in your video are bad for ORB-SLAM. Please, try in another area.
Also, to get good IMU initialization you need to tilt the system in pitch and roll at the beginning.

dinimar, 2020-10-20:
@jdtardos Thanks a lot for your reply. I'll try in another area, but in fact I've tested ORB-SLAM2(Stereo, no IMU) at the same place and such situation doesn't happen. Can you explain why do I need tilt in pitch and roll at the beginning?

jdtardos, 2020-10-20:
IMU initialization needs to estimate gravity direction and accelerometer bias. If you don’t tilt the IMU, both estimations get mixed, and this will produce drift in position.

leo-wkd, 2020-10-21:
hi, could you please share your D435i calibration yaml file? I am also using D435i to run stereo inertial package of ORB-SLAM3 but I found my calibration was different from what the program wants and therefore I cannot even initialize the tracking mode.

dinimar, 2020-10-29:
The reason of problem can be related to #132