UZ-SLAMLab/ORB_SLAM3 #154: [Crying] What is the IMU coordinate inside the code?

ns15417, 2020-10-14:
Thanks for your work, but i really gonna cry.... it is too painful which makes me think i am doing something stupid.
I feel really terrible now because of my unfinished work. i already spend two days to do the calibration. 
I am using realsense D435i and doing calibration by Kalibr, and got the result Tbc like this:
   data: [0.99986678 ,0.0002502 ,-0.0163204  ,-0.0182401 ,
         -0.00036682 ,0.99997442, -0.00714285,-0.00296991,
          0.0163182  ,0.00714789, 0.9998413  ,-0.0332756 ,
          0., 0., 0., 1. ]
it is matched with my imu data on with the gravity value is on imu y axis, like this:
`dx,dy,dz,rx,ry,rz,t -1.703988 -9.702253 0.166943 -0.036431 0.012424 -0.003512 1602666844.150768`

This Tbc clearly not ORBSLAM3 want, cuz after adding IMU datas, my keyFrames all turned to another direction which lead to Lost then.  And also i saw gz = (0,0,-9.8) which also makes me think that ORBSLAM3 has an certain coordinate for imu; so can anyone tell me what is wrong with my process or result?

leo-wkd, 2020-10-21:
same problem. I am also using D435i and Kalibr tools. The program initializes almost every second because of tracking lost or no  IMU measures.

d3637042, 2020-11-05:
I'm also testing out with D435i camera and also encountered the same issue of keyframe turning in some direction after initialization.
Did some research through the coordinates of example datasets (euroc & tum), camera frame is also defined as image below
[image]
The camera to imu(bodyframe) setups for the datasets
euroc:
[image]
https://projects.asl.ethz.ch/datasets/lib/exe/detail.php?id=kmavvisualinertialdatasets&media=platform.jpg
Tbc matrix from Examples/Monocular-Inertial/EuRoC.yaml
[0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975,
0.999557249008, 0.0149672133247, 0.025715529948, -0.064676986768,
-0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949,
0.0, 0.0, 0.0, 1.0]

tum:
[image]
https://vision.in.tum.de/data/datasets/visual-inertial-dataset
[-0.9995250378696743, 0.0075019185074052044, -0.02989013031643309, 0.045574835649698026, 
0.029615343885863205, -0.03439736061393144, -0.998969345370175, -0.071161801837997044,
-0.008522328211654736, -0.9993800792498829, 0.03415885127385616, -0.044681254117144367,
0.0, 0.0, 0.0, 1.0]

Both have correct transformation from the IMU to camera, so I believe the coordinates for defined in ORBSLAM should be like the first image. Don't know if I am missing out anything.

Phz-1995, 2020-12-13:
> same problem. I am also using D435i and Kalibr tools. The program initializes almost every second because of tracking lost or no IMU measures.

Hello, have you solved your problem? I encountered same problem as yours. I would  appreciate if you could help, thanks!

Phz-1995, 2020-12-13:
> I'm also testing out with D435i camera and also encountered the same issue of keyframe turning in some direction after initialization.
> Did some research through the coordinates of example datasets [...]

Hello friend, I'm also using D435i to test ORBSLAM3, according to the imu data from ROS wrapper I got I think the coordinate of D435i is different from both two above. It looks as:
[image]
I'm not sure if my result is right so could please share your opinion? By the way, have you successful run ORBSLAM3 with D435i? Looking forward to your reply!

muskie82, 2021-01-11:
Does anyone make it? I also would try Mono-Inertial with d435i, but no one seems to succeed.

alandszhang, 2021-02-17:
I am facing the same problem using D435i. Does anyone solve it?

Viky397, 2021-03-17:
Same problem for me, with the D435i.

jdtardos, 2021-12-23:
Version 1.0 now includes documentation of the reference systems, and examples to run ORB-SLAM3 with Intel Realsense cameras

closed 2021-12-23 by jdtardos