UZ-SLAMLab/ORB_SLAM3 #155: [Stereo-Inertial] Series of problems when mlQueueImuData is empty in Tracking::PreintegrateIMU()

zc08, 2020-10-15:
Thank you so much for sharing the code!

I'm trying to run the stereo-inertial system on my own dataset. However I've encountered a problem:
In Tracking.cc line 1115, function `void Tracking::PreintegrateIMU()`, it seems that when `mlQueueImuData` is empty, the function dirrectly returns, and the following assignments (beginning from line 1221) for mCurrentFrame are not implemented.

`    mCurrentFrame.mpImuPreintegratedFrame = pImuPreintegratedFromLastFrame;`   
`        mCurrentFrame.mpImuPreintegrated = mpImuPreintegratedFromLastKF;`   
`        mCurrentFrame.mpLastKeyFrame = mpLastKeyFrame;`

This seems to lead to a series of NULL pointer access in subsequent routines such as PredictStateIMU(), PoseInertialOptimizationLastFrame(), PoseInertialOptimizationLastKeyFrame(), etc.

I'm wondering, under which situation the `mlQueueImuData` could be empty? Is that normal or is there something wrong with my dataset? Thank you!

wangq0688, 2020-10-22:
hi, I have the same problem.
After IMU Initialization success, seg fault happens.

linyayunyayun, 2020-11-25:
hi, I have the same problem. The program crashed after a log "Not IMU data in mlQueueImuData!!".
Have you solved it?

wangq0688, 2020-11-25:
#162

Seasons2020, 2021-04-01:
Hellow ,I also meet this issue? Did you have sovled it?