UZ-SLAMLab/ORB_SLAM3 #157: The influence of 'Camera.fps' in config file in time performance of tracking

Phz-1995, 2020-10-15:
Hello everyone,

I'm using ORB_SLAM3 to study the **influence of parameters in config file** in the performance of **mono ORB_SLAM.** When I change the parameter 'Camera.fps', I get a confusing result:  **The lower the fps, the lower the time of ORB feature extracting.** In the code I can't find any relationship between them, so I'd like to know if someone know the reason, thanks!

Here is a screenshot of two tests on my own dataset, in the left test **30 fps** is used , in the right test **2 fps** is used

[image]

richard-elvira, 2020-10-20:
Hello, the fps in the config file only is relevant to the keyframe insertion criterium, in the case of 2 fps, at least a new keyframe is created for each 2 frames. The amount of keyframes is similar because the LocalMapping algorithm detect keyframes with redundant information and erase them from the map.

The extraction of features is not deterministic, the same sequence can obtain different times and features in the same frame, but it scales with the amount of features and the image size.

The Tracking time can be different depend of the local map, it scales with the size of the map points in the local map. The local map is deffined with the map points in the covisible and temporal keyframes close to the previous keyframe.

Phz-1995, 2020-10-21:
> Hello, the fps in the config file only is relevant to the keyframe insertion criterium, in the case of 2 fps, at least a new keyframe is created for each 2 frames. The amount of keyframes is similar [...]

Hi, Richard. Thank you very much for the reply.  I think all you said is right and helpful to understand  the principle of SLAM, but according to my tests I think the phenomenon, **that the extraction time of ORB features grows up with the fps may be not occasional**， I paste some tests results of two of my own datasets:

[image]
[image]

Looking forward to your help, thanks!

laxnpander, 2020-11-08:
@Phz-1995 I am not part of the ORB SLAM team, but your test is not representative. You will need to implement a much more elaborate test to prove your point. There can be several reasons why extraction time varies in your example. One could be background processes, or cpu load or just plain chance. Just like @richard-elvira said, it’s not deterministic. So you have to average your timings over a lot of frames to get a reasonable result. And even then your hardware might still be the reason and there is no such effect on other systems.