UZ-SLAMLab/ORB_SLAM3 #158: Mono + IMU on KITTI clips

liortalker, 2020-10-15:
Did anyone try to run with mono + imu on KITTI clips?
I get results with bad scales (among other inaccuracies).

There is some remark in the paper that for outdoor sequences the some IMU parameters diverge:
"In contrast, in some long outdoors sequences, the scarcity of close visual features may cause divergence of the inertial parameters, notably scale and accelerometer bias, which leads to errors in the order of 10 to 70 meters."

Not sure what is the relationship between far points and the accuracy of IMU parameters. Tell me if you have an idea about this.

cgringmuth, 2020-10-15:
Actually I'm currently trying to make Stereo + inertial work on our data (outdoor drive with a car). It is very bad. Stereo alone works quite well and is very robust, despite some drift which accumulates significantly after 2-3km. Introducing IMU data makes it really worse. Lost tracks after 5-10 sec. Turning despite driving straight and vice versa. I add a visualization of the IMU data to the map and compared it to the EuRoC data set where Stereo+IMU worked quite well. In both cases the IMU visualization looks very similar. So, no clue here what might be wrong.

I'm still in the process of figuring out if the IMU data I'm providing has errors or IMU configuration (e.g. transform) is not correct. But it looks good.

liortalker, 2020-10-15:
Thanks for your comment! 

For me stereo also worked very well for several datasets.
I'll share any information if I figure it out eventually.

jdtardos, 2020-10-19:
The IMU initialization requires to estimate accelerometer biases, separating them from gravity. For this, you need a motion with acceleration, and tilting the IMU with respect to gravity  in roll and pitch, which is quite difficult on a car. The best thing to do would be changing the initialization assuming zero accelerometer biases. This would work if your biases are very small. Otherwise, you need to research a better initialization method.

closed 2020-10-19 by jdtardos

liortalker, 2020-10-19:
Thanks! I will try.

renwuli, 2022-10-26:
@liortalker hi, have you tried to drop accelerometer bias estimation? What's the result?