UZ-SLAMLab/ORB_SLAM3 #166: Running ROS VI example on NVIDIA Jetson Xavier

PattyLiang, 2020-10-26:
Hello, I'm trying to run ROS example, the visual-only or RGBD are working, but whenever I test VI example I get the same output as #96 mentioned:

```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 458.654
- fy: 457.296
- cx: 367.215
- cy: 248.375
- k1: -0.283408
- k2: 0.0739591
- p1: 0.00019359
- p2: 1.76187e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
double free or corruption (out)
Aborted (core  #dumped)
```

The abort error is from ```Tracking.cc``` line 794: ```Tbc = node.mat();```
```
Thread 1 "Mono_Inertial" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x0000007fb6d344d8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x0000007fb6d358b4 in __GI_abort () at abort.c:79
#2  0x0000007fb6d6e65c in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fb6e2fb88 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x0000007fb6d749c4 in malloc_printerr (str=str@entry=0x7fb6e2b858 "double free or corruption (out)") at malloc.c:5350
#4  0x0000007fb6d765d8 in _int_free (av=0x7fb6e55a70 <main_arena>, p=0x5555b05f90, have_lock=<optimized out>) at malloc.c:4278
#5  0x0000007fb719279c in cvReleaseMat () at /usr/local/lib/libopencv_core.so.3.4
#6  0x0000007fb731b788 in cv::read(cv::FileNode const&, cv::Mat&, cv::Mat const&) () at /usr/local/lib/libopencv_core.so.3.4
#7  0x0000007fb78d4b14 in ORB_SLAM3::Tracking::ParseIMUParamFile(cv::FileStorage&) ()
    at /data/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#8  0x0000007fb78e7b40 in ORB_SLAM3::Tracking::Tracking(ORB_SLAM3::System*, DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>*, ORB_SLAM3::FrameDrawer*, ORB_SLAM3::MapDrawer*, ORB_SLAM3::Atlas*, ORB_SLAM3::KeyFrameDatabase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
    at /data/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#9  0x0000007fb78cb568 in ORB_SLAM3::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM3::System::eSensor, bool, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /data/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#10 0x0000005555565314 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /data/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_mono_inertial.cc:92
```

My PC version works well with:
1. Ubuntu 16.04
2. ROS Kinetic
3. OpenCV 3.4.3
4. Eigen3 3.2.92

Jetson Xavier with:
1. Ubuntu 18.04
2. ROS Melodic
3. OpenCV 3.4.3
4. Eigen3 3.3.4

I have tried to build libraries without ```-march=native```, and lower version of Eigen3, there still no luck.
I also test the subfunction inside ```ORB_SLAM3::Tracking::ParseIMUParamFile``` with OpenCV only, it can work well, means the version of OpenCV should be OK. 
I have no idea what reason cause the system crash, is anybody meets the same problem and can give some sugesstion??

PattyLiang, 2020-10-27:
Nevermind, I used `FileNodeIterator` to replace the the function of `Tbc = node.mat()` and it can workaround.

closed 2020-10-27 by PattyLiang

moon142857, 2020-10-29:
heiheihei 
I solved the problem!
didn't listen to my mother
Tested with OpenCV 3.2.0.

longzhenren, 2020-12-10:
> heiheihei
> I solved the problem!
> didn't listen to my mother
> Tested with OpenCV 3.2.0.

So you mean use 3.2.0 or not?/cry

WadeFrank, 2021-03-24:
> heiheihei
> I solved the problem!
> didn't listen to my mother
> Tested with OpenCV 3.2.0.

let me see see
i hope it can work good!

caomengnb, 2021-04-19:
> > heiheihei
> > I solved the problem!
> > didn't listen to my mother
> > Tested with OpenCV 3.2.0.
> 
> let me see see
> i hope it can work good!

did you try it? can OpenCV 3.2.0 work? hope for your reply

Taeyoung96, 2021-07-16:
@PattyLiang Hi, I got the same error that you got.  

Can you explain more about  
> Nevermind, I used FileNodeIterator to replace the the function of Tbc = node.mat() and it can workaround.  

this mention?   

How could I fix the `Tbc = node.mat()` in `Tracking::ParseIMUParamFile`?

PattyLiang, 2021-07-19:
@Taeyoung96 Hi, in ORB-SLAM3 v0.4, I don't meet this problem anymore, but you can refer the following solution.
```
        // Tbc = node.mat();
        // fSettings["Tbc"] >> Tbc;
        cv::FileNodeIterator it = node.begin();
        int Tbc_rows = (int)(*it)["rows"];
        int Tbc_cols = (int)(*it)["cols"];
        Tbc = cv::Mat(Tbc_rows,Tbc_cols,CV_32FC1);
```

iandouglas96, 2021-08-07:
> @Taeyoung96 Hi, in ORB-SLAM3 v0.4, I don't meet this problem anymore, but you can refer the following solution.
>
> ```
>         // Tbc = node.mat();
>         // fSettings["Tbc"] >> Tbc; [...]

This isn't entirely complete, since this will just make Tbc contain random data from memory.  Creating a new iterator is also not necesssary.  This will work (assuming float data):

```
int Tbc_rows = static_cast<int>(node["rows"]);
int Tbc_cols = static_cast<int>(node["cols"]);
Tbc = cv::Mat(Tbc_rows, Tbc_cols, CV_32FC1);
node["data"].readRaw("f", Tbc.data, Tbc_rows*Tbc_cols);
```