UZ-SLAMLab/ORB_SLAM3 #167: Build on OSX get error

rocketman123456, 2020-10-26:
In file included from /Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/Thirdparty/DBoW2/DBoW2/BowVector.h:14:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/map:910:5: error: 
      static_assert failed due to requirement
      'is_same<std::__1::pair<ORB_SLAM3::KeyFrame *, g2o::Sim3>,
      std::__1::pair<ORB_SLAM3::KeyFrame *const, g2o::Sim3> >::value'
      "Allocator::value_type must be same type as value_type"
    static_assert((is_same<typename allocator_type::value_type, value_type>::value),
    ^              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1063:21: note: 
      in instantiation of template class 'std::__1::map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::__1::less<ORB_SLAM3::KeyFrame *>,
      Eigen::aligned_allocator<std::__1::pair<ORB_SLAM3::KeyFrame *, g2o::Sim3>
      > >' requested here
    KeyFrameAndPose CorrectedSim3, NonCorrectedSim3;
                    ^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1096:29: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
            NonCorrectedSim3[pKFi]=g2oSiw;
            ~~~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1102:39: error: 
      use of undeclared identifier 'mit'; did you mean 'min'?
        for(KeyFrameAndPose::iterator mit=CorrectedSim3.begin(), mend=Co...
                                      ^~~
                                      min
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2551:1: note: 
      'min' declared here
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1102:30: error: 
      no member named 'iterator' in 'std::__1::map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::__1::less<ORB_SLAM3::KeyFrame *>,
      Eigen::aligned_allocator<std::__1::pair<ORB_SLAM3::KeyFrame *, g2o::Sim3>
      > >'
        for(KeyFrameAndPose::iterator mit=CorrectedSim3.begin(), mend=Co...
            ~~~~~~~~~~~~~~~~~^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1102:92: error: 
      use of undeclared identifier 'mit'; did you mean 'min'?
  ...mit=CorrectedSim3.begin(), mend=CorrectedSim3.end(); mit!=mend; mit++)
                                                          ^~~
                                                          min
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2551:1: note: 
      'min' declared here
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1102:97: error: 
      use of undeclared identifier 'mend'
  ...mit=CorrectedSim3.begin(), mend=CorrectedSim3.end(); mit!=mend; mit++)
                                                               ^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1102:103: error: 
      use of undeclared identifier 'mit'
  ...mit=CorrectedSim3.begin(), mend=CorrectedSim3.end(); mit!=mend; mit++)
                                                                     ^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1104:30: error: 
      use of undeclared identifier 'mit'
            KeyFrame* pKFi = mit->first;
                             ^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1105:41: error: 
      use of undeclared identifier 'mit'
            g2o::Sim3 g2oCorrectedSiw = mit->second;
                                        ^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1283:20: warning: 
      incrementing expression of type bool is deprecated and incompatible with
      C++17 [-Wdeprecated-increment-bool]
        mnFullBAIdx++;
        ~~~~~~~~~~~^
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1461:19: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
    vCorrectedSim3[mpCurrentKF]=g2oCorrectedScw;
    ~~~~~~~~~~~~~~^~~~~~~~~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1462:22: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
    vNonCorrectedSim3[mpCurrentKF]=g2oNonCorrectedScw;
    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1491:30: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
            vNonCorrectedSim3[pKFi]=g2oSiw;
            ~~~~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1498:27: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
            vCorrectedSim3[pKFi]=g2oCorrectedSiw;
            ~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1533:70: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
  ...Eigen::Matrix3d Rcor = eigR.transpose()*vNonCorrectedSim3[pKFi].rotation...
                                             ~~~~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1548:51: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
        g2o::Sim3 g2oCorrectedSwi = vCorrectedSim3[pKFref].inverse();
                                    ~~~~~~~~~~~~~~^~~~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1549:57: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
        g2o::Sim3 g2oNonCorrectedSiw = vNonCorrectedSim3[pKFref];
                                       ~~~~~~~~~~~~~~~~~^~~~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1772:38: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
                    vNonCorrectedSim3[pKFi]=g2oSiw;
                    ~~~~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1779:35: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
                    vCorrectedSim3[pKFi]=g2oCorrectedSiw;
                    ~~~~~~~~~~~~~~^~~~~
/Users/developer/Program/ThirdPartyLibrary/ORB_SLAM3/src/LoopClosing.cc:1798:82: error: 
      no viable overloaded operator[] for type
      'ORB_SLAM3::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM3::KeyFrame *,
      g2o::Sim3, std::less<KeyFrame *>,
      Eigen::aligned_allocator<std::pair<KeyFrame *, g2o::Sim3> > >')
  ...Eigen::Matrix3d Rcor = eigR.transpose()*vNonCorrectedSim3[pKFi].rotation...
                                             ~~~~~~~~~~~~~~~~~^~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
16 warnings and 20 errors generated.
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [all] Error 2

can't get a solution

Sologala, 2020-10-31:
T<T,  I just got an error on ubuntu 20.04 same with you.

Sologala, 2020-10-31:
@rocketman123456 HI i just find some way to tackle this problem .

this issue may cause by high version of GCC, casualy GCC 9, 10.

to tackle this, you can edit the `loopClosing.h`.

```
using KeyFrameAndPose = map<KeyFrame *, g2o::Sim3, std::less<KeyFrame *>,
                                    Eigen::aligned_allocator<std::pair<KeyFrame *const, g2o::Sim3>>>;
```

hope it will help you.

rocketman123456, 2020-10-31:
yes, this works @Sologala

closed 2020-10-31 by rocketman123456