UZ-SLAMLab/ORB_SLAM3 #170: A bug when initialize the visual-imu external parameters.

353264241, 2020-10-28:
In Imu_Types.h line 486 to line 492 
`void Calib::Set(const cv::Mat &Tbc_, const float &ng, const float &na, const float &ngw, const float &naw)
{
    Tbc = Tbc_.clone();
    Tcb = cv::Mat::eye(4,4,CV_32F);
    Tcb.rowRange(0,3).colRange(0,3) = Tbc.rowRange(0,3).colRange(0,3).t();
    Tcb.rowRange(0,3).col(3) = -Tbc.rowRange(0,3).colRange(0,3).t()*Tbc.rowRange(0,3).col(3);
    Cov = cv::Mat::eye(6,6,CV_32F);`
when the function input Tbc, it use `rowRange()` function to set the Tcb value, however the rowRange() isn't a lvalue in c++ and the value will still be identity();

[image]

a feasible usage is as after:

 Tcb = cv::Mat::eye(4, 4, CV_32F);

  cv::Mat Rcb = Tbc.rowRange(0, 3).colRange(0, 3).t();
  Rcb.copyTo(Tcb.rowRange(0, 3).colRange(0, 3));
  cv::Mat tcb = -Tbc.rowRange(0, 3).colRange(0, 3).t() *
                Tbc.rowRange(0, 3).col(3).clone();
  tcb.copyTo(Tcb.rowRange(0, 3).col(3));

tiantiandabaojian, 2020-10-29:
There is no BUG here

353264241, 2020-10-30:
> There is no BUG here

Have you test it? I test it under opencv 3.2.0,  the `rowRange(0,3)` couldn't be used as a left value.

tiantiandabaojian, 2020-10-30:
> > There is no BUG here
> 
> Have you test it? I test it under opencv 3.2.0, the `rowRange(0,3)` couldn't be used as a left value.

Tbc = Tbc_.clone();
    Tcb = cv::Mat::eye(4, 4, CV_32F);
    Tcb.rowRange(0, 3).colRange(0, 3) = Tbc.rowRange(0, 3).colRange(0, 3).t();
    Tcb.rowRange(0, 3).col(3) = -Tbc.rowRange(0, 3).colRange(0, 3).t() * Tbc.rowRange(0, 3).col(3);
    std::cout<<"Tcb"<<std::endl<<Tcb<<std::endl;
老铁这里跟啥版本关系不大，我是这么试的，不知道是不是你说的问题，反正我这里是有值的，这代码里有大量类似于上面的操作，要是都没用我觉得他不可能跑起来

353264241, 2020-10-30:
那你看下Tcb前后值是多少,这样赋值根本没有赋进去,它代码中是左值和`copyTo`混合使用的