UZ-SLAMLab/ORB_SLAM3 #171: Multiple Cameras

cs574296, 2020-10-29:
New to Github, so excuse me if this isn't the right place to be asking, but I was wondering what it would take to have 2 cameras publish to the same map?  I'm using ROS, and it seems like with the ATLAS structure it should be straightforward to have 2+ cameras taking data, and have those cameras merging maps whenever there is overlap.  I'm trying to extract visual odometry data from the cameras, but would like them to be in the same frame.

Just so you have some background, using ORB_SLAM2 I have code that sets up 2 instances of ORB_SLAM2 and reports the odometry for each camera, and am transforming those odometrys into my robots frame.  The problem is that if the 2 map frames of ORB_SLAM2 are misaligned (by some unknown amount), then I can't trust the relative orientation between the two cameras, and the data isn't as useful.

richard-elvira, 2020-11-09:
ORBSLAM is built to work with a single agent but main task are in different threads, you can put the Tracking in each agent and have a single server working with the complete map, you will need to think about the comunication between server-agent.

closed 2020-12-18 by jdtardos

ProVival, 2022-03-26:
Were you able to get any progress? I am trying to accomplish something similar to you, but i don't have any experience in this area