UZ-SLAMLab/ORB_SLAM3 #172: World frame coordinate changes mid operation when IMU is involved

thien94, 2020-10-30:
Thank you very much for making the work public. I encounter the following issue when using the package, hope someone can help.

In Mono-Inertial or Stereo-Inertial mode, the world frame changes mid-operation (it seems to be right after `InertialOptimization` is completed according to the console output) as can be seen in the example below. The world frame visualization is done by adding `pangolin::glDrawAxis(5.0);` to `Viewer.cc`.
[image]

This problem is evident when I print out the `Tcw` value returned by the [`TrackMonocular`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L406). However, I don't see such behavior with Mono or Stereo mode.

I would like to pose two questions:
1. My guess is that the first world coordinate alignment correspond to the IMU initialization (z and scale alignment), but I can't figure out why are the subsequent alignments happening?
2. Is the returned `Tcw` value (from `TrackMonocular`, `TrackStereo` and `TrackRGBD`) recommended to be consumed by other systems that requires odometry from ORB-SLAM3? If it is, the inconsistency in the coordinate frame seems undesirable. If it is not, what would be the parameter that can be used?

Thank you greatly for your time.

richard-elvira, 2020-11-09:
When the map is initialized the world reference is set in the first keyframe. Later, IMU is initializated and the world reference is changed to aligned to it, Z axis, previously aligned with the view of the first keyframe, is aligned with respect to the gravity vector.

thien94, 2020-11-10:
Hi @richard-elvira, thank you very much for the information. 

I can understand that the world reference frame is realigned to IMU's z axis when the initialization completes, but why does the origin jump to a different location that seems to be totally random (not linked to any keyframes obtained up to that point and not consistent between runs)?

martinakos, 2020-11-17:
Maybe related to my issue #164

thien94, 2020-11-18:
Hi @martinakos , thank you for the info. I can confirm that the error "NOT INERTIAL LINK TO PREVIOUS FRAME!!!!" also appears in my tests, so it's possible that the two issues are one and the same.

1216621137, 2021-04-13:
So, if I want get the real-time estimated pose, what would be the parameter that can be used? Or, it is just impossible to get the right pose  in real time.

dimaxano, 2021-05-04:
Hi, @thien94 
I am reading the ORBSLAM3 paper and found this detail (on the screenshot) about the initialization scheme in ORBSLAM3. 
Look like a possible answer to your question
[image]

thien94, 2021-05-05:
@dimaxano Thank you very much for the info. I didn't catch that when reading the paper.

18851409110, 2021-08-20:
I have downloaded the code and tried to run with a mono camera but without imu information.There are some problems such as running speed is a little slow. So have you ever tried running with a real camera but not read the dataset.  Maybe the dataset is the point. And have you ever tried other dataset?

BTW, is it convenient to add your wechat or other social account?

mateosss, 2021-10-02:
> but why does the origin jump to a different location that seems to be totally random (not linked to any keyframes obtained up to that point and not consistent between runs)?

Did you get an answer for this?

AyaRabea, 2021-11-01:
> I have downloaded the code and tried to run with a mono camera but without imu information.There are some problems such as running speed is a little slow. So have you ever tried running with a real [...]

I have the same problem did you find any solution?

dantenoguera, 2022-06-26:
> Hi, @thien94 I am reading the ORBSLAM3 paper and found this detail (on the screenshot) about the initialization scheme in ORBSLAM3. Look like a possible answer to your question [image]

Didn't get what this implies. Was someone able find an explanation for the jump when IMU is involved?

kaitaotang, 2022-11-04:
Maybe, the scale changed when the second and third initialization step. Therefore, it caused the pose of world coordinate changed as showed in your gif.

Shang-cl, 2023-06-13:
I met the same problem, can anyone help me.