UZ-SLAMLab/ORB_SLAM3 #176: It looks like a bug, but  not sure,  in frame.cc and KannalaBrandt8.cc

tiantiandabaojian, 2020-11-04:
In Frame::UndistortKeyPoints() 
Both Pinhole and KannalaBrandt8  use cv::undistortPoints
why not use cv::fisheye::undistortPoints ?

Another in KannalaBrandt8::ReconstructWithTwoViews 
Originally input is the point after undistortion, why to correct again ？

richard-elvira, 2020-11-09:
We use cv::undistortPoints always because the undistorted points are only used in the Pinhole model, in the fisheye model works only with the distorted points.

We will clean the code to avoid this kind of situations in the future.

tiantiandabaojian, 2020-11-09:
> We use cv::undistortPoints always because the undistorted points are only used in the Pinhole model, in the fisheye model works only with the distorted points.
> 
> We will clean the code to avoid this kind of situations in the future.

if (mDistCoef.at<float>(0) == 0.0)
 {
        mvKeysUn = mvKeys;
        return;
 }
in the fisheye model "mDistCoef.at<float>(0) == 0.0" is True?
This explains that the code itself is OK

richard-elvira, 2020-11-09:
Yes, in the Kannala model is always true, because the distortion model doesn't use that variable.

closed 2020-11-09 by richard-elvira