UZ-SLAMLab/ORB_SLAM3 #183: build_ros.sh------Problem!   make: *** No targets specified and no makefile found. Stop.

xwspace, 2020-11-16:
$ ./build_ros.sh 
Building ROS nodes
mkdir: cannot create directory "build": File exists
[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from syntax check of ORB_SLAM3/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ModuleNotFoundError: No module named 'rospkg'
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:78 (message):
  [rosbuild] Syntax check of ORB_SLAM3/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/yjvs/workspace/Projects/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found. Stop.

\---------------------------------------------------------------------------------------------------------------------------\
maybe the same issue with https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/89#issue-689338032
but, add ROS_PACKAGE_PATH cannot solve the problem. "export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS"

does anyone have any idea to solve this problem?

Kanishk598, 2021-01-29:
After adding the `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS` to your .bashrc file, you need to **Run** your .bashrc file. One way to run it is to re-open the terminal (.bashrc file runs automatically whenever a new terminal is started). Then, in this fresh terminal, enter the command `./build_ros.sh` and it should compile and work normally.

notu97, 2021-02-09:
@Kanishk598 I tried your suggestion,,,, yet it didn't work.

richardXia7462, 2021-05-13:
The version of python is wrong, change the python soft link. Follow the instruction.
cd /usr/bin
sudo rm python
sudo ln -s python3.8 python
It work for me. If you can't still fix the problem, it is welcome to give me feedback.