UZ-SLAMLab/ORB_SLAM3 #186: Symbol lookup error when run ./Example/Monocular/mono_euroc

thanghv97, 2020-11-18:
Enviroment:
  Ubuntu 18.04LTS
  Opencv 3.4.1

error content:

`./Example/Monocular/mono_euroc: symbol lookup error: ../../lib/libORB_SLAM3.so: undefined symbol: _ZN5DBow24FORB1LE`

I run ./build.sh to make project and ensure that lib DBow2 and g20 are also linked from Thirdparty of ORB_SLAM3

I can't understand what causes.
please tell me solutions,anyone.
thanks.

zhangciiiii, 2021-04-22:
> Enviroment:
> Ubuntu 18.04LTS
> Opencv 3.4.1
>
> error content:
>
> `./Example/Monocular/mono_euroc: symbol lookup error: ../../lib/libORB_SLAM3.so: undefined symbol: _ZN5DBow24FORB1LE`
>
> I run ./build.sh to [...]

Hi, I met the same problem, have you solved it ?

dnaglucose, 2021-05-22:
u can  
`ldd -r libORB_SLAM3.so`
find:
libDBoW2.so =>  xxxxx/xxxxx/xxxxx/lib/libDBoW2.so
remove this libDBoW2.so  and make  again, can solved this problem

I meet this same problem just now and then try to solve it.

1:`ldd -r libORB_SLAM3.so`
find:	`libDBoW2.so => /home/smz/slam_demo/kimera_ros/devel/lib/libDBoW2.so (0x00007ff0e6332000)`
and `undefined symbol: _ZN5DBoW24FORB1LE	(./libORB_SLAM3.so)`

2:`gedit ~/.bashrc`  and remove ` source ~/slam_demo/kimera_ros/devel/setup.bash` 
3:cmake .. and  make  
4:solve this problem

zhangciiiii, 2021-05-22:
> u can
> `ldd -r libORB_SLAM3.so`
> find:
> libDBoW2.so => xxxxx/xxxxx/xxxxx/lib/libDBoW2.so
> remove this libDBoW2.so and make again, can solved this problem
>
> I meet this same problem just now and then try [...]

Thanks! : )