UZ-SLAMLab/ORB_SLAM3 #187: How to save pose for Monocular-Inertial in ORB SLAM 3?

francescas94, 2020-11-18:
Hi everybody! Can anyone please help me creating a topic to save pose in Orb Slam 3? I'm running the Mono-Inertial with a INTEL T265 sensor. I'm new to this world so I really appreciate if someone could guide me step by step. It's very important for me because I need it to write my master thesis. Thank you very much!

mdasari823, 2020-11-19:
I am also facing same issue. I used this function`SLAM.SaveTrajectoryEuRoC` to save trajectory. But it doesn't look it is saving the poses for the entire trajectory.

francescas94, 2020-11-19:
@mdasari823 Thank you for your prompt response! Can you explain me how I should use the function you suggested? Sorry for the question, but I'm completely new to this world. Thank you a lot! I hope someone can help us to solve definitely the problem

mdasari823, 2020-11-21:
You can call this function in the main function of `Examples/ROS/ORB_SLAM3/src/ros_mono.cc` at the end. It should save the trajectory like below.

`SLAM.SaveTrajectoryEuRoC(''filename.txt");`

It is saving the entire trajectory. I misinterpreted it because of the scale. But it is working fine.

francescas94, 2020-11-21:
@mdasari823 Great! It works!! Thank you so much for your help!

closed 2020-11-21 by francescas94

xinankejidaxueliangchao, 2020-12-18:
> You can call this function in the main function of `Examples/ROS/ORB_SLAM3/src/ros_mono.cc` at the end. It should save the trajectory like below.
> 
> `SLAM.SaveTrajectoryEuRoC(''filename.txt");`
> 
> It is saving the entire trajectory. I misinterpreted it because of the scale. But it is working fine.

Hello, I encountered the same problem. I want to know who the saved pose is relative to, because I saw that the pose of the same frame is different in CameraTrajectory.txt and KeyFrameTrajectory.txt

ThakurSarveshGit, 2024-02-13:
> You can call this function in the main function of `Examples/ROS/ORB_SLAM3/src/ros_mono.cc` at the end. It should save the trajectory like below.
> 
> `SLAM.SaveTrajectoryEuRoC(''filename.txt");`
> 
> It is saving the entire trajectory. I misinterpreted it because of the scale. But it is working fine.

Do you mean `pSLAM->SaveTrajectoryEuRoC("filename.txt");` ?

For me, it runs, but does not creates the file! Any idea what could be wrong?

arjunskumar, 2025-06-11:
@ThakurSarveshGit it can be found in `/home/username/.ros`