UZ-SLAMLab/ORB_SLAM3 #192: RGB-D in ROS: The camera( Realsense D435) pose detected by the system is often still drifting when I stop rotating the camera.

ChuanlinSun, 2020-11-24:
Thank you for all your efforts.
I'm facing an issue with ORB_SLAM3 now:
I use camera Realsense D435 in RGB-D mode of ORB_SLAM3 by ROS. When I try to rotate the camera, the system can successfully detect the rotation. However, after I stop the rotation, the camera pose showed in the viewer thread sometimes keeps rotating in a slow by constant speed. I don't know what causes this problem because I have successfully use this camera with the same parameters in ORB_SLAM2 and there is no such a problem. Do you or anyone else face the same problem?

longzhenren, 2020-12-03:
Hello!
I met the similar circumstance when using the T265 Camera, I move the camera to to the direction of Southeast, and when I stopped moving, the track continue to move at a high speed. At last, I solved this by changing the IMU parameters by doing a squareroot. 
For D435, I think you can do calibrate once more.
Hope this can help you!

janis8008, 2020-12-08:
Hello!
I think there is a problem with Imu framework and ORBB-SLAM3 should work with calibrated data from camera, but these are imu calibration parameters that worked best for me
`Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [1,  0, 0,0,
          0, -1, 0,0,
          0,  0,-1,0,
          0.0, 0.0, 0.0, 1.0]

# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 0.001 # 0.004 (VINS) # 0.00016 (TUM) # 0.00016    # rad/s^0.5 
IMU.NoiseAcc: 0.001 # 0.04 (VINS) # 0.0028 (TUM) # 0.0028     # m/s^1.5
IMU.GyroWalk: 0.001 # 0.000022 (VINS and TUM) rad/s^1.5
IMU.AccWalk: 0.001 # 0.0004 (VINS) # 0.00086 # 0.00086    # m/s^2.5
IMU.Frequency: 200

thFarPoints: 20.0`

jdtardos, 2021-12-23:
Version 1.0 now includes examples to run ORB-SLAM3 with Intel Realsense cameras.

closed 2021-12-23 by jdtardos