UZ-SLAMLab/ORB_SLAM3 #203: OpenCV Error:

longzhenren, 2020-12-03:
OpenCV Error: Assertion failed (type == _src2.type()) in scaleAdd, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/src/matmul.cpp, line 2466
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/src/matmul.cpp:2466: error: (-215) type == _src2.type() in function scaleAdd

Aborted (core dumped)

I ran it on NVIDIA Jetson TX2(ARM-based), The version of opencv is  3.2.0 installed through ROS-melodic.
ORB_SLAM3 crashed when I was moving my camera and sensor. Just before the crash , everything seemed to work well.
How to solve that? Should I re-install my OPENCV or change it to another version? Which version will work on TX2?

ange63, 2021-01-12:
Same problem here. It seems to randomly happens even if using the same dataset. On my side it crashes inside `bool Tracking::PredictStateIMU()` some seconds after start. The first time the code has to choose between case 1 `if(mbMapUpdated && mpLastKeyFrame)` and case 2 `else if(!mbMapUpdated)`, case 1 always works, and case 2 always crashes because `mLastFrame.mVw` is empty (type == 0), which triggers the assertion inside `operator+(.)` that ises `scaleAdd`.

yhy130531, 2021-03-06:
hi, i also the meet the same problem as you when I run ORB_SALM3 on the jetson xavier NX , are you solved this problem now? Looking forward to your reply, thank you!

longzhenren, 2021-03-08:
To be honest, I do not know which of the followings works finally.
humm……The problem seems to be solved after changing the version of Eigen to **3.2.10** and use g2o from their official github repo : https://github.com/RainerKuemmerle/g2o ,then change the directories of the header files in some source files.
OpenCV is still 3.2.0 from ROS-melodic.
Change the version of Eigen3, then build and install this full version of g2o. After that maybe you should modify the MakeFile(Same with that in the /Examples/ROS/ORB_SLAM3 dir) to manually relocate the location of that.
Pay attention to the differences between /usr/local/include and /usr/include . That is annoying.

crashooo, 2021-04-18:
Same problem......