UZ-SLAMLab/ORB_SLAM3 #204: ROS-Node export definitiv image poses

jonasmeyer1, 2020-12-03:
Hi, thanks for the great work.
I am using the ROS Stereo-Inertial Node to process recorded data (ROS-Bag). How can I export the final image poses at the end of the SLAM processing? Does anyone have a simple solution?

janis8008, 2020-12-08:
Add
  ```
SLAM.Shutdown();

    // Save camera trajectory
  SLAM.SaveKeyFrameTrajectoryTUM("Results/results_key_frame.txt");
  SLAM.SaveTrajectoryEuRoC("Results/results_tra.txt");
  ros::shutdown();
```
To ROS example source code, rebuild examples, and after your .bag file ends do 
`rosnode kill -a` to stop all nodes and start saving trajectory data

jonasmeyer1, 2020-12-08:
Thanks for your answer. I just found the same solution in the code for the stereo-Node.

closed 2020-12-08 by jonasmeyer1