UZ-SLAMLab/ORB_SLAM3 #208: The RGBD SLAM perform worse than ORB SLAM2

ZhouXiner, 2020-12-08:
I try to run the RGBD type in TUM RGBD Dataset, but the performance is even worse than ORBSLAM2 in some sequence like fr1_desk2. Has anyone struggled the same problem like me?

janis8008, 2020-12-10:
How is it worse? 
Only changes in code compared to ORB-SLAM2 are :
`ORB_SLAM3`
`CV_LOAD_IMAGE`
And the difference in the Slam system is added imu support.
I don't think they changed the way how no Imu systems work.

ZhouXiner, 2020-12-10:
> How is it worse?
> Only changes in code compared to ORB-SLAM2 are :
> `ORB_SLAM3`
> `CV_LOAD_IMAGE`
> And the difference in the Slam system is added imu support.
> I don't think they changed the way how no Imu systems work.

Thanks for reply.
The APE is bigger in TUM RGBD dataset like fr1_desk2 and fr1_room.
Actually, the loop closuing part also changed a lot.
Maybe I should compare the preformance without loop.

melhashash, 2021-01-30:
I am experiencing the same issue with the monocular pipeline; the accuracy is worse than ORB-SLAM2.

CSU-NXY, 2021-03-24:
I've tested ORB-SLAM2 and ORB-SLAM3 on KITTI dataset in RGBD mode with my deep learning model predicted depth. It turns out that ORB-SLAM3 performs worse than ORB-SLAM2 and have scale drift issues.

CSU-NXY, 2021-03-24:
Hi, @ZhouXiner @melhashash Try this solution #107. It seems the RGBD mode didn't set mbf value correctly.

ZhouXiner, 2021-03-24:
> Hi, @ZhouXiner @melhashash Try this solution #107. It seems the RGBD mode didn't set mbf value correctly.

Thanks, I will try it.

melhashash, 2021-03-26:
> Hi, @ZhouXiner @melhashash Try this solution #107. It seems the RGBD mode didn't set mbf value correctly.

Thanks for your suggestion, but I am talking about the monocular case.

thomCastillo, 2021-07-31:
Could we add depth sensor error as parameter(i.e noise parameter in IMU) to optimization in RGB-D SLAM? If so, how can we add it?

UcefMountacer, 2021-09-17:
@ZhouXiner can you give me the link to dataset and the command? I tried on freiburg desk dataset but it doesn't work for me.

jdtardos, 2021-12-23:
Version v1.0 has corrected some bugs in RGB-D SLAM

closed 2021-12-23 by jdtardos