UZ-SLAMLab/ORB_SLAM3 #210: Running ORB-SLAM3 with a lower IMU update rate

p-fj, 2020-12-09:
Dear authors,
first of all, I am very impressed by your implementation and grateful that you share it with the community! Great work!

I was wondering whether it were possible to get acceptable results if the the IMU update rate was lower than 200Hz, let's assume 20Hz. 

In order to check that, I pruned the IMU readings in the data.csv and *.txt-file of the respective example of the EuRoC data set. I run into the following error, though

`./euroc_examples_MH5.sh 
Launching MH05 with Monocular-Inertial sensor
num_seq = 1
file name: dataset-MH05_monoi
Loading images for sequence 0...LOADED!
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stod
./euroc_examples_MH5.sh: line 10: 24861 Aborted                 (core dumped) ./Examples/Monocular-Inertial/mono_inertial_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular-Inertial/EuRoC.yaml "$pathDatasetEuroc"/MH05 ./Examples/Monocular-Inertial/EuRoC_TimeStamps/MH05.txt dataset-MH05_monoi
`

Am I missing to change a file or what could be the reason for the error?

And what is your expectation on the performance in low frequency IMU modes? 

Thank you very much! All the best, muchas gracias

janis8008, 2020-12-10:
Hello.
I have run ORB_SLAM3 with different Imu update rates, but I did it in realtime adjusting imu speed in camera config.
There should be an option to run the camera and Imu in sync, you should check it.

p-fj, 2020-12-10:
Hi Janis, 
thank you for your reply! Yes, indeed there are *.yaml-files (EuRoC.yaml and /mav0/imu0/sensor.yaml) where the rate can be set. This, however, does not have any effect on the imu frequency used as far as I can tell. I believe, it is the actual number of entries in the data.csv file and the respective timestamps which is relevant.

janis8008, 2020-12-11:
Hello,
I was talking about hardware clock speed, how often data is written in .bag, and another format data files.

ccamposm, 2020-12-15:
For EuRoC dataset, IMU readings are loaded from /mav0/imu0/data.csv. You should uniformly sample these measurements to get the desired frequency. Maybe, if the IMU rate is equal to fps there may exist no IMU measurements between consecutive frames, which can lead to some problems. I would suggest that you try first with some higher frequency, for example, 100 Hz, and see if this error persists. Notice there exist calibration parameters that are frequency-dependent (i.e. gyro and accelerometer noises and random walks) thus you will need to recompute them.

We have only tested with IMU rates over 100 Hz. In contrast with cameras, where increasing fps has a big impact on real-time performance, increasing IMU does significantly augment the computational burden. I will suggest that you use the highest IMU rate available for your device.

jdtardos, 2020-12-18:
Sorry, we mean increasing IMU does *NOT* significantly augment the computational burden

closed 2020-12-18 by jdtardos