UZ-SLAMLab/ORB_SLAM3 #211: Running  success but no response?!

RyanMiao11, 2020-12-10:
I run it success.but it failed  with this...
./Examples/Monocular-Inertial/mono_inertial_euroc /home/zj/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/zj/ORB_SLAM3/Examples/Monocular-Inertial/EuRoC.yaml /home/zj/EuRoc/MH01_easy  /home/zj/ORB_SLAM3/Examples/Monocular-Inertial/EuRoC_TimeStamps/MH01.txt
num_seq = 1
Loading images for sequence 0...LOADED!

janis8008, 2020-12-10:
Check if you have provided the right directories. It won't continue if it can't find files.

`        string pathCam0 = pathSeq + "/mav0/cam0/data";
        string pathImu = pathSeq + "/mav0/imu0/data.csv";`

Check if your final directories contain these files.

RyanMiao11, 2020-12-10:
> Check if you have provided the right directories. It won't continue if it can't find files.
> 
> ` string pathCam0 = pathSeq + "/mav0/cam0/data"; string pathImu = pathSeq + "/mav0/imu0/data.csv";`
> 
> Check if your final directories contain these files.

yes, you are right .thanks

closed 2020-12-10 by RyanMiao11