UZ-SLAMLab/ORB_SLAM3 #215: Any suggestions for choosing IMU?

XinnWang, 2020-12-17:
I am trying to find a good IMU for ORB_SLAM3. Which IMUs have you tried and how is their performance? Do you have any suggestions for IMU brand or type or price or performance or anything? Thanks so much.

AnhTuDo1998, 2021-01-11:
For Finaly Year Project in my university, some of my juniors used Intel Realsense d435i, maybe you can try that out. Personally, I have only worked on Realsense d435 for ORB-SLAM2 only

XinnWang, 2021-01-11:
@AnhTuDo1998 Thanks! Intel Realsense d435i is a depth camera with imu, but my project needs discrete imu modules. i will check the imu module inside d435i.

AnhTuDo1998, 2021-01-11:
Oh really sorry, i think i did not read it clearly

closed 2025-02-10 by XinnWang

khshmt, 2025-02-10:
I used VectorNav vn-100 for a project at work it is industrial grade IMU and have a sync out Signal to trigger other sensors like camera for example and can also deliver filtered Attitude