UZ-SLAMLab/ORB_SLAM3 #219: Deleting KeyFrames and MapPoints

KathiXIII, 2020-12-17:
The functions `Map::EraseMapPoint` and `Map::EraseKeyFrame` delete the pointer to map points and keyframes, but not the object themselves. Thus, when running ORB-SLAM3 for a big input file, the memory consumption gets a problem. I cannot solve it by erasing key frames and map points, since this does not free any memory. 
In the function a comment mentions "TODO: This only erase the pointer. Delete the MapPoint".
This was already an issue in ORB-SLAM2 (see https://github.com/raulmur/ORB_SLAM2/issues/350).
How can I safely delete keyframes and map points? Just adding `delete pMP;` and delete `pKF;` results in segmentation faults.

zhhclamber, 2021-01-06:
Have you solved it?

fishmarch, 2023-07-12:
I just solved this by using shared_ptr, [here](https://github.com/fishmarch/ORB-SLAM3-Remove-Bad). Hope it work for you.