UZ-SLAMLab/ORB_SLAM3 #221: Number of matches in Stereo vs Stereo-Inertial Tracking

mdasari823, 2020-12-19:
Hi, 

I am using Intel real sense T265 camera for Stereo and Stereo+Inertial tracking. I see a large gap in the number of matches for the same trajectory when I do Stereo vs Stereo+Inertial. While in Stereo mode, I see almost 300 matches in every frame, whereas in Stereo+IMU mode, I see less than 100 matches. Did anyone face this issue? Any reason why?

Also, it seems the tracking is really bad. I attached the two results below for your reference. Please let me know if you find any insights. 

[Stereo](https://drive.google.com/file/d/1ZHYAu_F8pjXCjq1ex6rs0JdkncN4hKVR/view?usp=sharing), [Stereo+Inertial](https://drive.google.com/file/d/10MjQjiXWi1jTCat7t9hSWjEoIw3YgV-W/view?usp=sharing), [Bag](https://drive.google.com/file/d/1rAtJyrMEDdEbQSm6BW4_d_32j0_mxj7c/view?usp=sharing)

ccamposm, 2020-12-20:
When running a visual-inertial system, two sources of errors usually exist:
1. IMU calibration. This includes intrinsic parameters (gyroscope and accelerometer noises and random walks densities) and also IMU to left camera transformation. These calibration parameters may be found using libraries like kalibr (https://github.com/ethz-asl/kalibr) and should be included in the .yaml calibration file (see /Examples/Stereo-Inertial/TUM_512.yaml).
2. Initial motion: The first two seconds of motion are used to find inertial parameters. The more aggressive (without losing visual track) the easier is to estimate them. Examples of good motions for IMU initialization may be found at TUM_VI dataset.

mdasari823, 2020-12-20:
Thank you @ccamposm for your quick response. I will check the IMU calibration and motion more carefully. 

But, I am curious why the number of matches in the case of visual-inertial is way smaller compared to just the visual system. Is there any particular reason for this?

BVM97, 2021-03-19:
Hi @mdasari823,

Have you made any progress? I am running into the same problem. VI seems to work fine until the robot starts turning. Then the number of features is close to none and OS3 looses the map.

mdasari823, 2021-03-19:
@BVM97 I did not understand their algorithm on why the number of matches were down when adding IMU, but I managed the problem by increasing the number of ORB features in the yaml file. That made sure the number of matches in case of VIO not to go very low and the tracking was fine.

antithing, 2021-06-13:
Hi @mdasari823, would you be able to please share your cpp and yaml for Stereo slam using the t265? Thank you!

qianyi93, 2021-06-22:
Hi@mdasari823  Have you figured out the reasons? I tested the mono and mono-inertial on the same dataset and encountered the same problem.

qianyi93, 2021-06-22:
Hi @mdasari823 Have you figured out the reasons? I tested the mono and mono-inertial on the same dataset and encountered the same problem.