UZ-SLAMLab/ORB_SLAM3 #225: Using ORB-SLAM3 with stereo camera in gazebo

kazuma-takeuchi-jpn, 2020-12-23:
Hi!
I'm trying to run ORB-SLAM3 with stereo camera in gazebo. I use ros melodic to do these things. Stereo camera works well, and ros topic about left and right camera is in the rostopic list. However, when I try to run ORB-SLAM3, it subscribe only left camera topic, in other words, only the image on the left side of the camera will be displayed.

[image]

This is the simulation environment.

[image]

This is the screen shot when I tried to run ORB-SLAM3

Does anyone know how to solve this problems? I thank you in advance, and Merry Christmas!

closed 2021-01-11 by kazuma-takeuchi-jpn

prasanna229, 2023-03-08:
hey did you slove the isssue...cz can you help me with it?

ababayi, 2023-07-13:
Hi, Can I access to your code?