UZ-SLAMLab/ORB_SLAM3 #228: Do any one solve the bug of SaveAtlas() and LoadAtlas()!!!

HappyXY, 2021-01-04:
i want to save the map and reuse the map for localization, there are some errors:
"Starting to write the save binary file
libc++abi.dylib: terminating with uncaught exception of type boost::archive::archive_exception: unregistered class - derived class not registered or exported "

if any one can save and load map successfully, please share the methods!
Thanks very much !!!

cntzj, 2021-03-24:
Have you solved this problem?

xinruozhishui201314, 2021-04-15:
我也在解决这个问题，要不大家加一下联系方式吧，可以更方便的沟通，我微信号：15517920165

xinruozhishui201314, 2021-04-17:
我現在能正常保存，加載還有點問題，在解決。

xinruozhishui201314, 2021-04-17:
@HappyXY 這個已經解決

neo-knight-td, 2021-04-19:
Could you provide you modifications ? @xinruozhishui201314

jdtardos, 2021-12-23:
Version v1.0 includes map loading and saving operations

closed 2021-12-23 by jdtardos