UZ-SLAMLab/ORB_SLAM3 #231: Very few matches in Stereo SLAM with T265 Fisheye camera

mdasari823, 2021-01-07:
Hi, I am using Intel Realsense T265 camera for Stereo SLAM. I am getting an extremely smaller number of matches as shown in the below figure. I am guessing this could be because of incorrect calibration or camara.bf parameters. Can you please check the following calibration file and let me know if I am missing anything. Thanks. 

Fisheye calibration file: 
[t265.yaml](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5778969/t265.txt)

[image]

I also tried it out using a PinHole model. The number of matches is pretty good (more than 200) but the tracking is really bad after turns. The calibration file is 
[here](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5778981/t265.txt)

mdasari823, 2021-01-10:
With proper camera calibration parameters, it seems the number of matches is fine even with the Fisheye camera model. But, there is still some Scale problem which seems to be off by a meter. Given this is stereo, I didn't expect any Scale problems. Is there a way I can correct the Scale?

aravindSolteq, 2021-04-23:
Hi Is the calibration provided by the Intel works??

mdasari823, 2021-04-23:
No, I had to use the Kalibr code to calibrate. The default was not working well.

aravindSolteq, 2021-04-23:
Is it possible to share your final yaml file? Also have you tried the stereo inertial SLAM?

antithing, 2021-06-04:
Hi @mdasari823 would it be possible for you to share your config file and the cpp that you use to run the t265 camera live?

Thank you!